libimobiledevice/libimobiledevice #1038: IOS14 USB tethering not working

luochongjun, 2020-09-24:
I used Tethering on Openwrt and it worked fine with IOS13.
Recently, my device has been upgraded to IOS14, and it doesn't work properly. 
I can successfully obtain the IP address 172.20.10.2 and ping 172.20.10.1, but I can't access the Internet, nor can I ping other IP addresses.
I checked openwrt's firewall and routing settings, and I can make sure they work.
At the same time, I was testing on Ubuntu 16.04.12 and had the same problem.
I also tried to compile the latest libimobiledevice and usbbmuxd source code, did not solve the problem.
I wrote a post on the openwrt forum, and others had the same problem
https://forum.openwrt.org/t/has-the-ios14-tethering-issue-been-resolved/74949

nikias, 2020-09-24:
Can you test this on macOS by any chance? Maybe it needs some permission or similar. Something must have changed, but also before this was not related to code in libimobiledevice. The code is implemented in the kernel here: https://github.com/torvalds/linux/blob/master/drivers/net/usb/ipheth.c
The fact that you get an IP address and can ping the phone shows that the kernel driver works though...

luochongjun, 2020-09-24:
Thanks,
Doesn't the usbmuxd  have any relevant code?

phrichuk, 2020-09-26:
I am having this same problem, and it is being mentioned now in more and more forums. I have not seen a message from anyone who has iOS14 iPhone USB tethering working on Linux/OpenWRT, so I believe it is currently broken.
I can see from page https://libimobiledevice.org/status/ "This list is based on known services until iOS 13.5." - so iOS14 has not yet been formally checked against. I wonder if it would be worth updating, to confirm this is currently broken for iOS14 - it might save time for people hunting round for a solution that does not appear to exist yet. Fingers crossed it is a 'quick fix' to mend this, and Apple have not done anything nasty in iOS14 to sabotage the great work that is libimobiledevice!

kkota1, 2020-09-26:
Note that this doesn't fix the issue, but I found that since updating to iOS 14, I now have to manually run `idevicepair pair` in order to get the "Trust this device?" dialog on the phone. IIRC, it used to automatically do this after the physical connection was detected.

Additionally, I can ping the iPhone gateway directly
```
traceroute to 172.20.10.1 (172.20.10.1), 30 hops max, 60 byte packets
 1  _gateway (172.20.10.1)  0.895 ms  0.842 ms  0.813 ms
```

but pinging google hangs.

```
PING google.com(lax31s11-in-x0e.1e100.net (2607:f8b0:4007:811::200e)) 56 data bytes
```

This suggests to me that there's an issue with iOS reading the packet and deciding not to forward it to the carrier. Note that DNS works.

I've tried this networkd config with and without masquerading to no avail

```
[Match]
Name=enp55s0u1c4i2

[Network]
DHCP=yes
IPMasquerade=yes
```

USB tethering still works on Windows 10, so I am going to attempt to compare the outgoing packets and spoof a W10 device.

nikias, 2020-09-26:
If you have to run `idevicepair pair` then something is wrong with the usbmuxd setup. usbmuxd is handling the pairing as soon as a device is detected, and this has the dialog pop up on the device. Regarding tethering, the only thing that usbmuxd does is that it selects a different USB config for the device. There is nothing done in particular regarding the tethering, the code to expose the USB ethernet device is handled by the ipheth kernel module. Maybe the setup sequence changed. Currently it sends a setup packet here:
https://github.com/torvalds/linux/blob/master/drivers/net/usb/ipheth.c#L250
We probably need a USB packet analyzer on a supported platform (macOS/Windows) to figure out what has changed or what we are missing.

phrichuk, 2020-09-26:
Thanks for this. I am getting a bit old now, my era was ZX Spectrum and BBC Micro... I would have had a go myself at solving equivalent problems back in the day... glad to see the younger generation (I assume!) getting stuck into this problem!
In terms of symptoms, I had not noticed any recent change in the 'trust device' dialogue behaviour myself. I can ping the gateway as above, and agree DNS seems to work. I upgraded to iOS 14.0.1 earlier today, and the issue remains as per iOS14. And it works fine for me too on W10. Something must have changed within iOS since v13.x- there are several traceroute steps within the iPhone between 172.20.10.1 and the outside world when this is working (why???), so something must have changed in there somewhere! This must be solvable - because it still works on W10! Therefore best of luck to kkota1 and others trying to spot the significant difference between W10 behaviour and Linux/WRT behaviour here, then hopefully emulating the 'new' required behaviour via Linux/OpenWRT as per W10!

kkota1, 2020-09-26:
Packet comparison for usb tethering, google.com ping request, seq = 1. Linux above (no ping response), Windows below (success).

[image]

lwhsu, 2020-09-26:
@kkota1 thanks for the analysis and experiments. I got totally the same issue here. The packets you captured look like the ethernet packets on the network interface. I think as @nikias suggested, we may need to compare the USB packets used to setup tethering. Maybe this helps https://gitlab.com/wireshark/wireshark/-/wikis/CaptureSetup/USB .

kkota1, 2020-09-26:
@lwhsu ok I will try to do that

nikias, 2020-09-26:
@kkota1 nice, could you expand the ipv6 entries or maybe even upload the pcap files?

kkota1, 2020-09-26:
@nikias attached
[packet-compare.zip](https://github.com/libimobiledevice/libimobiledevice/files/5287271/packet-compare.zip)

kkota1, 2020-09-26:
@nikias I attached the raw usb packets from successful windows pings. See time 4.9 to 8.5s

I won't have time to try the linux usb packets today but hopefully someone else here is able to

[windows-ping-usbpcap.zip](https://github.com/libimobiledevice/libimobiledevice/files/5287295/windows-ping-usbpcap.zip)

nikias, 2020-09-26:
@kkota1 thanks for the pcap files. I was looking on mobile before, so I didn't see it's just the ICMP packets. I think the problem here will be the actual "handshake" that occurs when the device is attached to the host. This is assuming that the actual network device works fine (and it seems to do, since ping to the phone works, and apparently DNS also). Something is preventing the traffic from leaving the phone, so it feels like there's a new "pairing" sequence or similar that needs to be done for this to work.

kkota1, 2020-09-26:
If I am reading this correctly, when I replayed the windows traffic from linux, it worked. So this is indeed a packet issue? This is my first time using tcpreplay

```
[kiran@kiran-linux packet-compare]$ sudo tcpreplay -i enp55s0u1c4i2 -K hotspot-windows.pcapng 
[sudo] password for kiran: 
File Cache is enabled
Actual: 329 packets (158214 bytes) sent in 21.73 seconds
Rated: 7277.9 Bps, 0.058 Mbps, 15.13 pps
Flows: 15 flows, 0.69 fps, 115 flow packets, 214 non-flow
Statistics for network device: enp55s0u1c4i2
	Successful packets:        329
	Failed packets:            0
	Truncated packets:         0
	Retried packets (ENOBUFS): 0
	Retried packets (EAGAIN):  0
```

nikias, 2020-09-26:
Hmm does the ICMP ping get a reply back when replying? Maybe it is something completely different, like some ipv6 issue

kkota1, 2020-09-26:
I am seeing successful ping responses but I'm not sure if it's just replaying both request AND responses on my local machine, or if it's actually sending the requests out and waiting for real responses from the internet.

I am currently trying tcpliveplay instead

luochongjun, 2020-09-27:
I compared the USB packets of WIN10 and Linux, and found that **feature configuration packets** will be sent to iPhone in WIN10, but Linux does not. I am trying to set it in Linux.

luochongjun, 2020-09-27:
> If I am reading this correctly, when I replayed the windows traffic from linux, it worked. So this is indeed a packet issue? This is my first time using tcpreplay
>
> ```
> [kiran@kiran-linux [...]
Very cool, I will try

luochongjun, 2020-09-27:
[image]
For the SET Feature USB data packets captured on Win10, the same type of data packets do not appear on Linux

luochongjun, 2020-09-27:
Attach the usb package
[usb package.zip](https://github.com/libimobiledevice/libimobiledevice/files/5287631/usb.package.zip)

luochongjun, 2020-09-27:
I found that every time I got the MAC address in Win10, the following packets would be sent out. From the description, they should be specific to iPhone.
[image]
I try to simulate this packet in Linux, but I don't know how to do it. I can simulate setup data, but I can't simulate URB Function. Who can give me some advice?
[image]

nikias, 2020-09-27:
Good find. The dump shows the "CARRIER" packet
```
{bmRequestType: 0xc0, bRequest: 69, wValue: 0x0000, wIndex: 0, wLength: 1}
```
which returnes `04` in the buffer.

Then the above packet:
```
{bmRequestType: 0xc1, bRequest: 3, wValue: 0x0000, wIndex: 2, wLength: 4}
```
Returns `02 00 00 00` in the buffer. Directly after there's also another one:
```
{bmRequestType: 0xc1, bRequest: 3, wValue: 0x0000, wIndex: 2, wLength: 4}
```
Returns `00 00 00 00` in the buffer.

Finally it shows a "get mac address" request
```
{bmRequestType: 0xc0, bRequest: 0, wValue: 0x0000, vIndex: 2, wLength: 6}
```
The "CARRIER" and "get mac address" packets are handled by ipheth; so it seems we are missing the two new ones there.

nikias, 2020-09-27:
Right... there's also that OUT URB
```
{bmRequestType: 0x40, bRequest: 64, wValue: 0x0064, wIndex: 0, wLength: 0}
```

luochongjun, 2020-09-27:
Both CHECK_CARRIER and GET_MAC get the correct return values in Linux, but it's hard to determine if the packet is of request type bmRequestType: 0x40. I didn't find a set URB Function in the driver.

nikias, 2020-09-27:
This would send the OUT URB after getting the mac address. Maybe that solves it?
```
--- a/ipheth.c	2020-09-27 14:51:03.000000000 +0200
+++ b/ipheth.c	2020-09-27 14:55:27.000000000 +0200
@@ -284,6 +284,7 @@
 	struct usb_device *udev = dev->udev;
 	struct net_device *net = dev->net;
 	int retval;
+	int retval2;

 	retval = usb_control_msg(udev,
 				 usb_rcvctrlpipe(udev, IPHETH_CTRL_ENDP),
@@ -307,6 +308,20 @@
 		retval = 0;
 	}

+	retval2 = usb_control_msg(udev,
+			usb_rcvctrlpipe(udev, IPHETH_CTRL_ENDP),
+			0x40, /* request */
+			0x40, /* request type */
+			0x64, /* value */
+			0x00, /* index */
+			NULL,
+			0,
+			IPHETH_CTRL_TIMEOUT);
+	if (retval2 < 0) {
+		dev_err(&dev->intf->dev, "%s: usb_control_msg: %d\n",
+			__func__, retval2);
+	}
+
 	return retval;
 }
```

nikias, 2020-09-27:
One thing I also noticed is that ipheth is sending the CARRIER packet every second. Maybe the device doesn't like that either.

kkota1, 2020-09-27:
I submitted the bug info and this thread to the /net/usb maintainers' email

I don't have the expertise to help with the bug fix but if anyone wants more pcap files or me to do QA work, just let me know. Thanks all

luochongjun, 2020-09-28:
I've added all the different packets to Linux, but the problem remains.
I had to try to find official support for IOS.

kkota1, 2020-10-02:
The Linux maintainers messaged me back that they probably can't help with this.

Note about Windows 10: It will not work unless iTunes is installed and "iTunes Mobile Device Helper" service is running. This service used to be clearly displayed in the services list, but *puts on tin foil hat*, it seems like Apple+Microsoft have gone to some lengths to hide it. You have to enable it in the W10 settings GUI or launch iTunes for it to run. Should mention that iTunes is now a Microsoft Store app, not a third party .exe

Here is a diff of the services that came up after I launched iTunes.

OneSyncSvc_919ca is the only one without the Win32OwnProcess flag. I think Microsoft/Apple bundled their pairing process into one of these.

Name | DisplayName
fdPHost | Function Discovery Provider Host
FDResPub | Function Discovery Resource Publication
FontCache | Windows Font Cache Service
jhi_service | Intel(R) Dynamic Application Loader Host Interface Service
lmhosts | TCP/IP NetBIOS Helper
LMS | Intel(R) Management and Security Application Local Management Service
OneSyncSvc_919ca | Sync Host_919ca
SgrmBroker | System Guard Runtime Monitor Broker
StorSvc | Storage Service
UsoSvc | Update Orchestrator Service
wscsvc | Security Center
wuauserv | Windows Update

Only solution may be to switch to Android if they really want to keep this proprietary as I suspect

nikias, 2020-10-02:
Well I am quite confident this can be made operational. Maybe we are just looking at the wrong place...

kkota1, 2020-10-03:
I agree that it can be made operational. But it might be wise to wait until 14.1 or 14.2 before you spend your time.

Reasoning: Apple changed a perfectly operational interface and didn't make the new one backwards compatible. They obfuscated the Windows side so much. These could be discounted as rookie mistakes if they weren't coming from Apple.

My guess is it's to make it more difficult for third party hardware manufacturers to integrate with them so that they can sell their name brand. Linux users aren't their target market so they can deal with the fallout. In that case, who's to say they won't break it in iOS 15 and again in 16 to dissuade us third parties.

I hope I'm wrong, but if I'm right we are just giving them free red teaming services by trying to crack the interface.

kkota1, 2020-10-03:
[win-handshake.zip](https://github.com/libimobiledevice/libimobiledevice/files/5320878/win-handshake.zip)

Here's another pcap from windows. Pairing starts at 26 seconds

corsac-s, 2020-10-07:
Hi, I experienced the issue as well, and I'll try to check with the kernel people on how to move forward. @nikias could you share the patch you tried for a start?

Is there some kind of contact with Apple which could be activated to get some information about the protocol? Or analysing what's done on MacOS/Windows in the only way?

corsac-s, 2020-10-11:
In my case, on Linux, here's the observed traffic when enabling Personal Hotspot *while already plugged in*

```
ipheth -> iphone: {bmRequestType: 0xc0, bRequest: 69, wValue: 0x0000, wIndex: 2, wLength: 64}
iphone -> ipheth: 040403
ipheth -> iphone: {bmRequestType: 0xc0, bRequest: 69, wValue: 0x0000, wIndex: 2, wLength: 64}
iphone -> ipheth: 040403
```
When disabling i get:

```
ipheth -> iphone: {bmRequestType: 0xc0, bRequest: 69, wValue: 0x0000, wIndex: 2, wLength: 64}
iphone -> ipheth: 030303
```

nikias, 2020-10-11:
Yes 04 means the hotspot connection is available. The ipheth implementation tries to read 64 bytes, while the original reads just 1.

corsac-s, 2020-10-11:
Ack. I don't have a way to monitor usb traffic of a Windows or MacOS host, but what messages are emitted when enabling personal hotspot on an already plugged device?

Could you share the patch you already tried (besides the one already pasted above but which only has one new message)?

corsac-s, 2020-10-12:
So I tried the below quick and very dirty patch and like @nikias it didn't work. But I noticed something from the Windows capture which might be relevant.
In that capture there's a SET INTERFACE call, with interface=2 and altInterface=2, while ipheth has a call with interface=2 and altInterface=1. So maybe there's something fishy as well here.

I also tried to ping @dgiago (writer of the original [code](https://github.com/dgiagio/ipheth)) by mail but didn't get an answer yet.

```
diff --git a/drivers/net/usb/ipheth.c b/drivers/net/usb/ipheth.c
index b09b45382faf..5e35b78bb248 100644
--- a/drivers/net/usb/ipheth.c
+++ b/drivers/net/usb/ipheth.c
@@ -337,6 +337,22 @@ static int ipheth_open(struct net_device *net)
 
 	usb_set_interface(udev, IPHETH_INTFNUM, IPHETH_ALT_INTFNUM);
 
+	dev_err(&dev->intf->dev, "%s: sending %d request\n", __func__, 0x40);
+	retval = usb_control_msg(udev,
+			usb_sndctrlpipe(udev, IPHETH_CTRL_ENDP),
+			0x40, /* request */
+			0x40, /* request type */
+			0x64, /* value */
+			0x00, /* index */
+			NULL,
+			0,
+			IPHETH_CTRL_TIMEOUT);
+	if (retval < 0) {
+		dev_err(&dev->intf->dev, "%s: usb_control_msg: %d\n",
+			__func__, retval);
+	}
+	dev_err(&dev->intf->dev, "%s: done sending %d request\n", __func__, 0x40);
+
 	retval = ipheth_carrier_set(dev);
 	if (retval)
 		return retval;
@@ -477,10 +493,61 @@ static int ipheth_probe(struct usb_interface *intf,
 		goto err_alloc_ctrl_buf;
 	}
 
+	dev_err(&dev->intf->dev, "%s: sending %d request\n", __func__, 0x03);
+	retval = usb_control_msg(udev,
+			usb_sndctrlpipe(udev, IPHETH_CTRL_ENDP),
+			0x03, /* request */
+			0xc1, /* request type */
+			0x00, /* value */
+			0x02, /* index */
+			dev->ctrl_buf, IPHETH_CTRL_BUF_SIZE,
+			IPHETH_CTRL_TIMEOUT);
+	if (retval < 0) {
+		dev_err(&dev->intf->dev, "%s: usb_control_msg: %d\n",
+			__func__, retval);
+	} else if (retval < 4) {
+		dev_err(&dev->intf->dev,
+			"%s: usb_control_msg: short packet: %d bytes\n",
+			__func__, retval);
+		retval = -EINVAL;
+		goto err;
+	} else {
+		dev_err(&dev->intf->dev,
+			"%s: usb_control_msg: %d response %phN\n",
+			__func__, 0x3, dev->ctrl_buf);
+	}
+
+	dev_err(&dev->intf->dev, "%s: sending %d request\n", __func__, 0x02);
+	retval = usb_control_msg(udev,
+			usb_sndctrlpipe(udev, IPHETH_CTRL_ENDP),
+			0x02, /* request */
+			0xc1, /* request type */
+			0x00, /* value */
+			0x02, /* index */
+			dev->ctrl_buf, IPHETH_CTRL_BUF_SIZE,
+			IPHETH_CTRL_TIMEOUT);
+	if (retval < 0) {
+		dev_err(&dev->intf->dev, "%s: usb_control_msg: %d\n",
+			__func__, retval);
+	} else if (retval < 4) {
+		dev_err(&dev->intf->dev,
+			"%s: usb_control_msg: short packet: %d bytes\n",
+			__func__, retval);
+		retval = -EINVAL;
+		goto err;
+	} else {
+		dev_err(&dev->intf->dev,
+			"%s: usb_control_msg: %d response %phN\n",
+			__func__, 0x2, dev->ctrl_buf);
+	}
+
 	retval = ipheth_get_macaddr(dev);
 	if (retval)
 		goto err_get_macaddr;
 
+
+err:
+
 	INIT_DELAYED_WORK(&dev->carrier_work, ipheth_carrier_check_work);
 
 	retval = ipheth_alloc_urbs(dev);
```

corsac-s, 2020-10-12:
Another data point, still trying to debug this. On an unpatched Linux kernel, inserting the ipheth module shows that line in idevicesyslog

```Oct 12 20:48:32 kernel(IOAccessoryPortUSB)[0] <Notice>: IOAccessoryPortUSB::controlRequest unknown request 0x0 bmReqType 0xc0 configured(0,0)```

Which I assume is related to the `IPHETH_CMD_CARRIER_CHECK` requests. Not sure if this one should not be done, or if we do it out of sync.

nikias, 2020-10-12:
Hmm the syslog message seems to be pointing to bmReqType 0xc0 with request 0x0 which is the "get mac address" request. Not sure why this would cause an 'unknown request' since we also found this request in the windows dump...

corsac-s, 2020-10-13:
Yes indeed, but maybe we sent it at a wrong time or something

jgcaap, 2020-10-18:
I see you guys already developping a solution. Keep up the good work. :)

AbbasBHF, 2020-10-20:
Hi, I'm an amateur in Linux and I have a Debian just because windows is very slow on my pc.
I have IOS 14.1 USB tethering problem on my iPhone SE.
I have an idea but I don't know if it's working or not.
How IOS find out that I connected from Linux?
There is a way to say to IOS that this OS is not Linux and it's windows?

shafzhr, 2020-10-20:
> Hi, I'm an amateur in Linux and I have a Debian just because windows is very slow on my pc.
> I have IOS 14.1 USB tethering problem on my iPhone SE.
> I have an idea but I don't know if it's working or [...]

It's not the problem. The problem is the intial communication that changed in IOS 14.

boolves, 2020-10-20:
Praying for a fix soon. I badly need USB tether to do my work!

boolves, 2020-10-20:
Here is a dmesg if this helps

`[   44.120750] usbcore: registered new device driver apple-mfi-fastcharge
[   44.170678] ipheth 1-1.1:4.2: Apple iPhone USB Ethernet device attached
[   44.170726] usbcore: registered new interface driver ipheth
[   44.185568] ipheth 1-1.1:4.2 enp0s26u1u1c4i2: renamed from eth0
[   45.799055] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s26u1u1c4i2: link becomes ready
[   45.920280] audit: type=1130 audit(1603236669.147:85): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   56.011866] audit: type=1131 audit(1603236679.241:86): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   59.196990] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:86:dd SRC=2a01:04f8:0172:1d86:0000:0000:0000:0001 DST=2600:0380:7528:15a4:1850:3f74:0f55:8b0d LEN=80 TC=88 HOPLIMIT=61 FLOWLBL=155821 PROTO=TCP SPT=80 DPT=50122 WINDOW=27160 RES=0x00 ACK SYN URGP=0 
[   62.538018] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:08:00 SRC=40.126.5.33 DST=172.20.10.5 LEN=60 TOS=0x18 PREC=0x40 TTL=61 ID=0 PROTO=TCP SPT=443 DPT=59408 WINDOW=27560 RES=0x00 ACK SYN URGP=0 
[   62.986340] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:08:00 SRC=138.201.81.199 DST=172.20.10.5 LEN=60 TOS=0x18 PREC=0x40 TTL=61 ID=0 PROTO=TCP SPT=80 DPT=40552 WINDOW=27560 RES=0x00 ACK SYN URGP=0 
[   63.195859] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:86:dd SRC=2a01:04f8:0172:1d86:0000:0000:0000:0001 DST=2600:0380:7528:15a4:1850:3f74:0f55:8b0d LEN=80 TC=88 HOPLIMIT=61 FLOWLBL=962275 PROTO=TCP SPT=80 DPT=50122 WINDOW=27160 RES=0x00 ACK SYN URGP=0 
[   63.258572] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:86:dd SRC=2a01:04f8:0172:1d86:0000:0000:0000:0001 DST=2600:0380:7528:15a4:1850:3f74:0f55:8b0d LEN=80 TC=88 HOPLIMIT=61 FLOWLBL=510824 PROTO=TCP SPT=80 DPT=50122 WINDOW=27160 RES=0x00 ACK SYN URGP=0 
[   63.497757] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:08:00 SRC=40.126.5.33 DST=172.20.10.5 LEN=60 TOS=0x18 PREC=0x40 TTL=61 ID=0 PROTO=TCP SPT=443 DPT=59408 WINDOW=27560 RES=0x00 ACK SYN URGP=0 
[   63.978704] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:86:dd SRC=2a01:04f8:0172:1d86:0000:0000:0000:0001 DST=2600:0380:7528:15a4:1850:3f74:0f55:8b0d LEN=80 TC=88 HOPLIMIT=61 FLOWLBL=705639 PROTO=TCP SPT=80 DPT=50132 WINDOW=27160 RES=0x00 ACK SYN URGP=0 
[   64.133855] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:08:00 SRC=138.201.81.199 DST=172.20.10.5 LEN=60 TOS=0x18 PREC=0x40 TTL=61 ID=0 PROTO=TCP SPT=80 DPT=40552 WINDOW=27560 RES=0x00 ACK SYN URGP=0 
[   64.499979] [UFW BLOCK] IN=enp0s26u1u1c4i2 OUT= MAC=42:bc:60:89:08:6c:42:bc:60:98:ec:64:08:00 SRC=40.126.5.33 DST=172.20.10.5 LEN=60 TOS=0x18 PREC=0x40 TTL=61 ID=0 PROTO=TCP SPT=443 DPT=59408 WINDOW=27560 RES=0x00 ACK SYN URGP=0 
`

luochongjun, 2020-10-21:
> Praying for a fix soon. I badly need USB tether to do my work!

May not be able to solve it in a short time. You can connect to the hotspot through WIFI first.

boolves, 2020-10-21:
> May not be able to solve it in a short time. You can connect to the hotspot through WIFI first.

Unfortunately it's not as reliable as a connection. a lot of my factories don't have good internet connection or my train commutes I opt for USB-tether since my mobile provider is faster.

Do you know if there is a way to roll back iOS 13? not much is showing up online

nikias, 2020-10-21:
> > May not be able to solve it in a short time. You can connect to the hotspot through WIFI first.
> 
> Unfortunately it's not as reliable as a connection. a lot of my factories don't have good internet connection or my train commutes I opt for USB-tether since my mobile provider is faster.
This doesn't make much sense, the WiFi is between your phone and your laptop and uses the mobile data of the phone, just as it would via USB.
 
> Do you know if there is a way to roll back iOS 13? not much is showing up online
Apple doesn't sign iOS 13 anymore, so no.

jgcaap, 2020-10-21:
I think he meant thAt is faster then Wi-Fi. 

I also believe all of us should give time to the developer to figure out.

If you need anything from our side just leave a comment I am more than happy to help. 👍 
Good work

toazd, 2020-10-21:
Apologies ahead of time if I'm repeating information. I don't have the time right now to read this entire issue. I played around a little bit this morning and I have to leave very soon.

I'm not sure if this is helpful to anyone involved with solving or diagnosing this issue or not but just in case...

After tethering my phone (iOS 14.0.1) and then monitoring the created ethernet interface (enp*) using wireshark I can see that A and AAAA DNS queries _appear_ to be handled successfully (I have also tried a few other valid domains and they all _appear_ to work too).

In the following screenshot from Wireshark my phone is 172.20.10.1 and the monitoring PC is 172.20.10.3.
https://imgur.com/a/IoFuHYf

I can provide captures from both Linux and Windows 10 from iOS 14.0.1 and iOS 14.1 if it would be helpful to anyone diagnosing this issue (USB or otherwise).

rainlake, 2020-10-26:
Just to be clear. DNS query does not work same as normal traffic. If iPhone is running a DNS daemon like dnsmasq on the device it just same as you are able to ping the gateway.

olijf, 2020-10-29:
Apple release IOS 14.1 last week. Release notes mention something about compatibility with network equipment. Anybody get a chance to test it out already? I hope they fixed it already...

lwhsu, 2020-10-29:
> Apple release IOS 14.1 last week. Release notes mention something about compatibility with network equipment. Anybody get a chance to test it out already? I hope they fixed it already...

I've upgraded and tested, still broken.

olijf, 2020-10-29:
> > Apple release IOS 14.1 last week. Release notes mention something about compatibility with network equipment. Anybody get a chance to test it out already? I hope they fixed it already...
> 
> I've upgraded and tested, still broken.

Tnx, then I dont have to go looking for wifi to download the update.

bobhruska, 2020-10-29:
> After tethering my phone (iOS 14.0.1) and then monitoring the created ethernet interface (enp*) using wireshark I can see that DNS queries _appear_ to be handled successfully (I have also tried a few other valid domains and they all _appear_ to work too).

When the tether comes up, the IP address of the gateway (172.20.10.1) is always reachable.  It will not route traffic anywhere else, though.  That 172.20.10.1 also happens to be a DNS server, so DNS lookups will work.  You just can't route anywhere.

samsouk, 2020-11-02:
I have the same problem since the iOS 14.1 upgrade

* Tethering over USB: DNS lookup works, but can't route anywhere.
* Tethering over Bluetooth: works

j3hempsey, 2020-11-05:
Another interesting data point, using an iOS device with iOS 14.0.1 and attempting to use USB tethering with a Linux box.

Network setup looks like:
Linux client -> iPhone -> 4G internet -> Linux server

At the moment, I am currently unable to take captures on the Linux client, however taking tcpdumps on the Linux server and using cURL from the client, I see SYN's coming in from the Linux client (and am able to confirm that they are indeed from that specific client) and SYN-ACK's returning back to the client. So I know packets are traversing the network from the client to the server, and from the server back to the client, but not traversing the iPhone and making it back to the Linux client itself as the cURL never completes or progresses (I only see re-transmitted SYN's and SYN-ACK's on the server).

Attempting to send raw UDP packets from the Linux client to the server, I get no packets in the capture on the server (`echo "ASD" > /dev/udp/<server-ip>/9999`) which roughly indicates to me that those packets are not traversing the iPhone at all. 

I found this difference in traffic while attempting to debug this exact issue before stumbling upon this ticket. I know at this point you all have been seeing differences in the pairing setups between Windows and Linux boxes, and subsequently looking into USB issues, however I thought this might be an interesting data point to add.

DonToblerOhne, 2020-11-12:
Tested on IOS 14.2 (just released around the 6th of November 2020).

USB tethering on a Manjaro rig still doesn't seem to work. Pre-14 it was OK on the same machinery.

From the computer I'm able to ping my gateway (the IOS device). There also seems to be name resolution. But anything else than the gateway will not answer.

It's like NAT or packet forwarding doesn't perform as it should be.

Please let me know if you have a different experience than mine.

Thanks!

EmlJnke, 2020-11-12:
https://discussions.apple.com/thread/251829006?answerId=253622748022#253622748022

In the Apple Support forum folks are saying Synology already figured out the problem... Does that help in any way?

corsac-s, 2020-11-12:
> https://discussions.apple.com/thread/251829006?answerId=253622748022#253622748022
> 
> In the Apple Support forum folks are saying Synology already figured out the problem... Does that help in any way?

Hi, thanks for the pointer. If there's a Linux patch in Synology OS, then I guess we should be able to extract it and reinject it upstream. I'll take a look when possible.

[100 more replies]

closed 2020-12-28 by nikias