libimobiledevice/libimobiledevice #1039: need help for my "hello world"

yuanjifeng, 2020-09-25:
want to complete https://libimobiledevice.org/#example-content-connecting in my raspberrypi.
connect.c:
[image]
gcc error:
[image]

how to resolve ?

mexmer, 2020-09-25:
well that sample is wrong in first place. but if you have any experience with C coding, you should see it and fixit yourself.

nikias, 2020-09-25:
Line 12 must be `&device` and line 17 `&udid`
Not sure what happened with the markup on the page but we'll get it fixed. Also usually it should be
`#include <libimobiledevice/libimobiledevice.h>`

mexmer, 2020-09-25:
looks read sample has same problem @nikias

nikias, 2020-09-25:
Yes I saw that. We'll fix it.

yuanjifeng, 2020-09-25:
[image]
gcc error：
[image]

how to resolve ?

nikias, 2020-09-25:
I think you should learn basic things about compilation first before attempting to use this. You need to pass the libraries that you want to use:
```
gcc -o connect connect.c -limobiledevice
```

yuanjifeng, 2020-09-25:
gcc connect.c -limobiledevice-1.0
it works.
thank you very much.

closed 2020-09-25 by yuanjifeng

nikias, 2020-09-27:
FYI the samples on the website are now fixed.