libimobiledevice/libimobiledevice #1040: problem Compiled libimobiledevice for windows

brtvrdrl, 2020-09-25:
i have been trying for 10 days to compile libimobiledevice for windows but i couldnt achieve it. Can someone compile it for windows please. i really need help

bricedenice59, 2020-10-03:
You won't be able to compile libimobiledevice in its latest version (let's say with some source code from around May-June this year) as there is still an unresolved issue with a dependency (libusbmuxd)
 check this out : https://github.com/libimobiledevice/libusbmuxd/issues/95
I hope they can resolve that as this is blocking everybody trying to compile it on Windows, at least with mingw...

YSL21, 2020-11-11:
Compile seemed to work but still got the same issue as of yesterday... any updates on a fix?