libimobiledevice/libimobiledevice #1044: Install downloaded mobileconfig profile

karthikeyan1241997, 2020-10-07:
I downloaded mobileconfig profile via email.

Is there any command to install the downloaded mobileconfig profile instead of manually installing the profile via settings?

Orbif, 2020-10-13:
Look like there is a pull request adding this possibility, but I haven't tested it : #1029

a-gunter, 2020-10-13:
I haven't tried the PR either, but I can confirm that the MCInstall service works to install .mobileconfig profiles onto the device

saurav-997, 2020-10-20:
I also wanted to know the same thing, if there is any possibility then kindly tell.

ethanatwork, 2020-10-21:
Hey, just saw this. That's my pull request, and unfortunately, the answer is yes and no. Devices will accept certain configuration profiles installed before setup is completed on the device, but not after you have reached the home screen. If the device is still in setup you can use the tool in my PR call ideviceprofile to install a .mobileconfig and it will not require user confirmation. If the device is at the home screen the tool will behave the same as installing it by email, the user will have to confirm the install manually in settings. Hope this helps.

beckietech, 2021-08-13:
hi ethan, could you tell me which branch to check out for testing? Thank you in adv.

ethanatwork, 2021-08-28:
> hi ethan, could you tell me which branch to check out for testing? Thank you in adv.

https://github.com/libimobiledevice/libimobiledevice/pull/1029