libimobiledevice/libimobiledevice #1055: LOCKDOWN_E_SSL_ERROR == lockdownd_client_new_with_handshake

XanaduAPI, 2020-10-30:
When I trust the device, I use it  lockdownd_client_new_with_handshake Reconnection，Will appear LOCKDOWN_E_SSL_ERROR 。
But when I put userpref_save_pair_record（）Inside plist_to_bin（）Revised as plist_to_xml().There are fewer mistakes

XanaduAPI, 2020-10-30:
Windows Mingw32

mexmer, 2020-10-30:
lockdown_e_ssl_error appears when certificate is not recognized by device, that means it was generated incorrectly. are you using correct openssl version for that?

XanaduAPI, 2020-11-02:
@mexmer 
I use lockdown_ client_ new_ with_ Handshake connection, then set a timer, close it after 5 seconds, and use lockdown in a new process_ client_ new_ with_ Handshake is ready to connect. This is a success

XanaduAPI, 2020-12-02:
I use lockdownd_client_new_with_handshake() creates a client, UDID.plist The build is successful and has a value, but lockdownd_client_new_with_handshake()  in The lockdownd_start_session() call failed, and I found that PEM_read_bio_X509(membp, &rootCert, NULL, NULL); this line of code will fail to call.
Of course, not all devices have this problem.
Once this problem occurs, deleting the pairing file and reconnecting will not work.

What's more, when will the common/socket file be updated? Libusbmuxd can be compiled on windows.

@nikias

XanaduAPI, 2020-12-02:
@nikias 
If you can, I hope you can provide the version and compilation parameters of OpenSSL you are using.

XanaduAPI, 2020-12-02:
cd openssl-1.1.1
./config shared no-asm enable-ssl2 enable-ssl3 enable-tls -fPIC -DOPENSSL_PIC enable-weak-ssl-ciphers

I found that the above parameters can solve this problem

closed 2020-12-02 by XanaduAPI

mexmer, 2020-12-02:
@XanaduAPI you should not need SSL2, you need SSL3 if you want to connect devices with iOS < 4.0, as for weak-ciphers, i'm not sure what is currently considered weak in 1.1 so there migh be some issues with elder versions of iOS, but it's safe option to use.

XanaduAPI, 2020-12-02:
@mexmer   Mainly : enable-weak-ssl-ciphers
If it is not enabled, some devices may not be connected

mexmer, 2020-12-03:
yes, that's what i meant, while in iOS4 they added TLS 1.0 support, it does not support like 256bit ciphers, so using weak ciphers is required, i'm not sure about let's say, ios 10 and newer tho' i have ios 9 (ipad 2), and it requires weak ciphers too. while ios 13 and 14 work without weak ciphers.