libimobiledevice/libimobiledevice #1056: Related problems of multithreading support

XanaduAPI, 2020-11-02:
Can I ask if 'libimobiledevice' supports multithreaded concurrent requests? I tried to use multithreading to call, often failed, and the data was not what I expected.

mexmer, 2020-11-07:
it's not libimobile, but device, that doesn't support multithreading, you can have only single session with lockdown service, you can send multiple requests in single sessionyou can have different services running on device and read data from all. like start notification proxy, springboard and afc.

also all services can be started only once per device connection, some will autoclose when you disconnect from them, while some (like AFC) can be started only once, until device is disconnected and connected again or until you disconnect from service with explicit goodbye.

XanaduAPI, 2020-11-07:
@mexmer 
I start all services in the main thread, and then I distribute the handles to other threads. Sometimes a MUX_ERROR error occurs in multi-threaded calls. Is this concurrent access undesirable?

closed 2021-01-05 by XanaduAPI