libimobiledevice/libimobiledevice #1058: After successful backup, idevicebackup2 list fails and idevicebackup2 info hangs

Ben-Voris, 2020-11-03:
The backup command completes with "Backup Successful" but the list command fails and the info command hangs. My concern is that the backup is somehow corrupt.

My goal is to restore a backup of my iPhone6S Plus to my iPhone 12 Pro (whenever it arrives).

To test these tools, I started with an iPhone 5. 

ideviceinfo reports:

```
iPhone5,2
ProductVersion: 10.3.3
```

idevicebackup2 reports no version information, so I don't know what I have. Running on Ubuntu 18.04.

The backup command I used and the few lines of the backup output are:

```
: idevicebackup2 backup --full ./iphone5
Backup directory is "./iphone5"
Started "com.apple.mobilebackup2" service on port 50877.
Negotiated Protocol Version 2.1
Starting backup...
Enforcing full backup from device.
Backup will be unencrypted.
Requesting backup from device...
Full backup mode.
```

And the last few lines are:

```
[==================================================]  99% Finished
Moving 128 files
[==================================================]  99% Finished
Moving 128 files
[==================================================] 100% Finished
Moving 128 files
[==================================================] 100% Finished
Moving 128 files
[==================================================] 100% Finished
Moving 100 files
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
[==================================================] 100% Finished
Removing 1 file
[==================================================] 100% Finished
Removing 1 file
[==================================================] 100% Finished
[==================================================] 100% Finished
Sending '7365f22cb55feb57a030ba076fb8499dd1170603/Status.plist' (189 Bytes)
Sending '7365f22cb55feb57a030ba076fb8499dd1170603/Manifest.plist' (25.9 KB)
Sending '7365f22cb55feb57a030ba076fb8499dd1170603/Manifest.db' (7.5 MB)
Received 3699 files from device.
Backup Successful.
```

(I'm creating an unencrypted backup because of this issue [idevicebackup2 info command not working on iOS 11 encrypted backup](https://github.com/libimobiledevice/libimobiledevice/issues/607). Even though this phone is running iOS 10, the backup password is not accepted (but the same password was accepted by `idevicebackup2 encryption off`).

The command `idevicebackup2 list ./iphone5` prints this:

```
ErrorCode 205: Error reading backup database (MBErrorDomain/205). Underlying error: No such file or directory at path "7365f22cb55feb57a030ba076fb8499dd1170603/Manifest.mbdb" (MBErrorDomain/4)
```

Here are the files whose names begin with Manifest:

```
./iphone5/7365f22cb55feb57a030ba076fb8499dd1170603/Manifest.plist
./iphone5/7365f22cb55feb57a030ba076fb8499dd1170603/Manifest.db
```

nikias, 2020-11-03:
The list and info, and also the unback commands are broken. Apple broke it and they don't care to fix it.

Ben-Voris, 2020-11-06:
They broke these command even on old iOS versions?

Is there any way to verify a backup?

nikias, 2020-11-06:
They broke it some time with iOS 8 or 9 IIRC.

mexmer, 2020-11-07:
@nikias they broke unback around ios 8, and list/info in ios 10, when they switched from MDBD format to SQLite format for manifest database.

as for backup verification, there is not much one can do, you can use any tool for reading itunes backup, and see if everything is in place, but that's about it (eg. compare content of manifest database with content on disc)

nikias, 2020-11-07:
You can have a look at https://github.com/rickmark/libibackup which deals with the new backup format.

mexmer, 2020-11-09:
@nikias that one doesn't look like it supports encrypted backups. only unprotected backups.

nikias, 2020-11-09:
One of the tools has a check_files feature. It doesn't matter if encrypted or not if the only thing you care about is verifying that the backup integrity is okay.

mexmer, 2020-11-09:
in encrypted backup, manifest.db is encrypted, so you can't verify integrity, because you can't load database.

nikias, 2020-11-09:
Oh right... I didn't even see that yet.

Ben-Voris, 2020-11-10:
Do I understand that, for an encrypted backup, the only way to test the backup is to try to use it?

If that is correct, and if the backup is corrupt or incomplete, what problems could the restore cause? A bricked device? Loss of data?

mexmer, 2020-11-11:
what you cannot do is validate backup trough mobile device (which **idevicebackup2** tool does), because apple BROKE most of backup functionality inside phone ... **backup** and **restore** still work correctly, other commands not much.

@Ben-Voris no, you don't need to "Try it". There exists 3rd party programs, that allow backup browsing, just the one that @nikias linked does not support encrypted backup, but if you search google for iphone backup browser, you will get many hits either for paid or free programs.

Ben-Voris, 2020-11-11:
My concern is not browsing the backup but verifying that its contents
matches what is on the i-device. In other words, I'm looking for something
like the --compare function of tar.

mexmer, 2020-11-13:
there is no such function and never was in first place.
for file content verification you can use sha hash of file content, which is present in metabase.

maybe some backup browser on internet includes this function.

itscool900, 2025-12-18:
as of version 1.4.0 i have no such problem
@Ben-Voris do you have any open concerns now?