libimobiledevice/libimobiledevice #1060: Install App from AppStore

karthikeyan1241997, 2020-11-03:
Is there any tool or command to start installation of specific app from app store using bundle identifier?

nikias, 2020-11-03:
Unfortunately not.

nikias, 2020-11-03:
Only thing that comes to my mind is when restoring from backup you can place a file (and idevicebackup2 is doing that) so after the final reboot of the restore procedure it will start downloading the previously installed apps as specified by that file. The code that handles that you can find here: https://github.com/libimobiledevice/libimobiledevice/blob/bfaf1a132a0d528cfe1dae8cbb4f0ecdeccaa88d/tools/idevicebackup2.c#L543

Adderally, 2021-05-31:
You can push and IPA on the phone, but it appears the device needs to be jailbroken .. i know 3uTools can install apps but you need to have an AppleID linked to the software