libimobiledevice/libimobiledevice #1061: Android Build

XanaduAPI, 2020-11-05:
How to compile on Android? For example, with NDK, maybe there should be one Android.mk Or Application.mk ?

leleliu008, 2020-11-17:
@XanaduAPI 

you can use [ndk-pkg](https://github.com/leleliu008/ndk-pkg) to build C/C++ project for Android.
```
ndk-pkg update
ndk-pkg install libimobiledevice --min-sdk-api-level=24
```

XanaduAPI, 2020-11-18:
> @XanaduAPI
> 
> you can use [ndk-pkg](https://github.com/leleliu008/ndk-pkg) to build C/C++ project for Android.
> 
> ```
> ndk-pkg update
> ndk-pkg install libimobiledevice --min-sdk-api-level=24
> ```

But I need to modify the source code to meet my needs

leleliu008, 2020-11-18:
@XanaduAPI 

you can edit the formula of libimobiledevice to meet your needs.

```
ndk-pkg formula edit libimobiledevice
```

XanaduAPI, 2020-11-19:
@leleliu008 
I want to modify libplist, libusbmuxd, libimobile device, libirecovery. Are these all ok

leleliu008, 2020-11-19:
@XanaduAPI   yes. `libplist` `libusbmuxd` `libimobiledevice` `libirecovery` all are ok.  It is very easy to use, you can give it a try.

XanaduAPI, 2020-11-30:
@leleliu008

XanaduAPI, 2020-11-30:
@leleliu008 你是北京的？

XanaduAPI, 2020-12-01:
You can compile as follows

sudo apt update
sudo apt upgrade
sudo apt install libtool
sudo apt install automake
sudo apt install autoconf
sudo apt install python
sudo apt install pkg-config
sudo apt install pkgconf

export ANDROID_HOME=/home/xanadu/Android
export ANDROID_ARM_32=$ANDROID_HOME/ARM_32
$NDK_HOME/build/tools/make-standalone-toolchain.sh --platform=android-24 --arch=arm --install-dir=$ANDROID_ARM_32
export PKG_CONFIG_PATH=$ANDROID_ARM_64/lib/pkgconfig
export PATH=$ANDROID_ARM_64:$PATH
export PATH=$PKG_CONFIG_PATH:$PATH
export PATH=$ANDROID_ARM_64/bin:$PATH

libplist
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 --without-cython

libusbmuxd
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64

openssl
./Configure --cross-compile-prefix=aarch64-linux-android- --prefix=$ANDROID_ARM_64 linux-armv4 no-asm no-engine no-ui

libimobiledevice
./autogen.sh --host=aarch64-linux-android --prefix=$ANDROID_ARM_64 --without-cython

closed 2020-12-01 by XanaduAPI

daomeixiong2547, 2021-03-23:
> libplist
> ./autogen.sh --host=arm-linux-androideabi --prefix=$ANDROID_ARM_32 --without-cython
> libusbmuxd
> ./autogen.sh --host=arm-linux-androideabi --prefix=$ANDROID_ARM_32 --without-cython
> openssl
> ./Configure --cross-compile-prefix=arm-linux-androideabi- --prefix=$ANDROID_ARM_32 linux-armv4 no-asm no-engine no-ui
> libimobiledevice
> ./autogen.sh --host=arm-linux-androideabi --prefix=$ANDROID_ARM_32 --without-cython

libplist libusbmuxd openssl libimobiledevice 四个指令执行顺序有区别吗？

leleliu008, 2021-07-15:
[image]

`openssl` and `libusbmuxd` is depended by `libimobiledevice`
`libplist` is depended by `libusbmuxd`

xanadu-lee, 2022-09-16:
@devops35 
First, you need to modify libusb so that it can load devices via the device path (eg: /dev/bus/usb/001/001).
Then compile usbmuxd and all other libraries into a dynamic library, which is called using Android JNI.

hcropps, 2022-09-17:
Thank you very much for your reply @xanadu-lee .

How can I change libusb. I can actually get the iphone usb path with usbmanager. Where should I use this usb path?

I also compiled all the necessary libraries with ndk-pkg. I included the so files from here into the project with jni. I wrote a binding c file myself. I can run the codes. But still usbmuxd is not working. Is usbmuxd not working because my android phone is not rooted?

I tried the https://github.com/cczhr/OTGLocation project. They have stated in the issues that this project will not work without root.

xanadu-lee, 2022-09-19:
@hcropps 
I mean, Root is not needed at all.
You have to modify the libusb code because libusb cannot enumerate devices on Android.
So you need to manually add the interface to tell libusb that new devices are added/removed.
Then when the device is opened, the device descriptor needs to be obtained from Android and returned to libusb.

Then you also need to modify libimobiledevice to pass in a readable and writable directory to store pairing files.

hcropps, 2022-09-25:
@xanadu-lee Thank you for your help. I've been struggling for weeks and still haven't been able to compile it for android. How can I talk to you privately?

xanadu-lee, 2022-09-26:
@hcropps 
I'm sorry I can't discuss this further with you.
Because some of the stuff involved belongs to the company and not to me.
All I can tell you is that the core problem lies in enabling libusb to notify the caller.
So you need to add some interfaces for libusb to pass Android data to libusb.
If you solve this one, then the others are small problems and will be solved easily.

For example add the following interface
libusb_android_device_add();
libusb_android_device_remove();

When you successfully solve this problem, both usbmuxd and libimobiledevice can successfully communicate with the device.

hcropps, 2022-09-26:
@xanadu-lee Thank you very much for your help. I have one last question for you. please help me. Libusb worked as you said. Now I can get device information via libusb. But when I run the idevice_get_device_list_extended function, it cannot find the device. I think the problem is related to usbmuxd. Do I need to run usbmuxd manually with code? Why can't it find the device?

hcropps, 2022-10-03:
@xanadu-lee I've been trying for weeks but I couldn't do it. Please help me a little more. I couldn't find how to start usbmuxd. I added it to android as bin file. When I copy it into /var/local/tmp I get permission denied error. If I call it in native lib then this time libimobiledevice-glue-1.0.so etc. can't find it. Where should I run usbmux? I'm trying so hard but I can't figure it out.

xanadu-lee, 2022-10-04:
@hcropps 
Do as I said last time.
Build all dependencies as lib(.a) libraries.
Then link all libraries in your own dll(.so).
Of course, you also need to build usbmuxd as a lib(exe >> lib) library.
This way they can share all dependencies.
Overall, there is only one so that is finally built and placed on Android. There is nothing else.

hcropps, 2022-10-10:
@xanadu-lee I tried to do what you said. I put usbmuxd in libs folder. I compiled all the so files together via jni. When I run libusb in c file it works but not when I run it via usbmuxd. So I send fileDescriptor via usbmuxd and try to init in usbmuxd but libusb init doesn't work. libusb_wrap_sys_device returns smaller than zero. I think it's permission problem. How can i solve this problem. Am I running it wrong?

// that's how I run it.
String command2 = myContext.getApplicationInfo().nativeLibraryDir + "/libusbmuxdreal.so -v -f --fileDescriptor "+Integer.toString(fd);
       Process proc2 = Runtime.getRuntime().exec(command2);

// I changed this file as follows.
usbmuxd/src/usb.c

libusb_context *ctx = NULL;
libusb_device_handle *devh = NULL;
int r = 0;
//r = libusb_set_option(NULL, LIBUSB_OPTION_NO_DEVICE_DISCOVERY, NULL);
r = libusb_set_option(&ctx, LIBUSB_OPTION_NO_DEVICE_DISCOVERY, NULL);
if (r != LIBUSB_SUCCESS) {
usbmuxd_log(LL_NOTICE, "libusb_set_option failed: %d\n", r);
return -1;
}
r = libusb_init(&ctx);
if (r < 0) {
usbmuxd_log(LL_NOTICE, "libusb_init failed: %d\n", r);
return r;
}
r = libusb_wrap_sys_device(ctx, (intptr_t)fileDescriptor, &devh);
if (r < 0) {
usbmuxd_log(LL_NOTICE, "libusb_wrap_sys_device failed 111: %d\n", r);
r = libusb_wrap_sys_device(NULL, (intptr_t)fileDescriptor, &devh);
if (r < 0) {
usbmuxd_log(LL_NOTICE, "libusb_wrap_sys_device failed 222: %d\n", r);
return r;
}

xanadu-lee, 2022-10-13:
@hcropps 
As I described before, when libusb opens usb, it needs Android to pass in the callback descriptor, and usbmuxd can't do anything for libusb.
You have to pass in a callback that is called whenever libusb tries to open the device, this callback executes the Java code that opens the device in Android and returns an available handle. libusb uses this handle for read and write operations.
Also, usbmuxd does not need to enumerate devices when initializing libusb.
The enumeration operation is performed by Android and notifies libusb.

hcropps, 2022-10-13:
@xanadu-lee 
I made the libusb related parts in my jni c file. I am able to init libusb and get device information. But when I run usbmuxd it also tries to init libusb and throws an error. There are parts that I still don't fully understand. Below is what I did. Could you please tell me where I went wrong?

1- I get fileDescriptor on java side with UsbManager.
2- I init libusb over jni c file with this fileDescriptor.
3- I run my usbmuxd.so file that I put under the libs folder with java (myContext.getApplicationInfo().nativeLibraryDir + "/libusbmuxdreal.so -v -f")

I still don't know how to set up the relationship between usbmuxd and libusb. I also changed only socket_paths on usbmuxd and libusbmuxd and made the application's files folder. What else do I need to do?

I know it's taking up a lot of your time, but I've been trying for weeks and still can't get it to work. Thank you very much for helping me. Can you share your e-mail address with me? My e-mail address: hicrandevapps@gmail.com

xanadu-lee, 2022-10-14:
@hcropps 
libusb_wrap_sys_device doesn't actually do anything useful.
Moreover, the file descriptor is libusb actively asking Java, not Java actively calling libusb.

davidwang5201, 2023-06-20:
@xanadu-lee 大佬，你集成的so文件成功的通过usb拿到apple设备数据了吗？比如说系统日志

xanadu-lee, 2023-06-21:
> @xanadu-lee 大佬，你集成的so文件成功的通过usb拿到apple设备数据了吗？比如说系统日志

可以拿到，并且所有操作均可以使用。

davidwang5201, 2023-06-21:
> > @xanadu-lee 大佬，你集成的so文件成功的通过usb拿到apple设备数据了吗？比如说系统日志
> 
> 可以拿到，并且所有操作均可以使用。

这个so文件怎么生成啊，使用上面的[ndk-pkg](https://github.com/leleliu008/ndk-pkg) 吗？
这个ndk-pkg要求在linux环境才可以使用...

xanadu-lee, 2023-06-21:
> > > @xanadu-lee 大佬，你集成的so文件成功的通过usb拿到apple设备数据了吗？比如说系统日志
> > 
> > 
> > 可以拿到，并且所有操作均可以使用。
> 
> 这个so文件怎么生成啊，使用上面的[ndk-pkg](https://github.com/leleliu008/ndk-pkg) 吗？ 这个ndk-pkg要求在linux环境才可以使用...

过程很繁琐，而且需要改源代码。
你如果不是很精通Android与C++，最好不要弄。

davidwang5201, 2023-06-21:
> > > > @xanadu-lee 大佬，你集成的so文件成功的通过usb拿到apple设备数据了吗？比如说系统日志
> > > 
> > > 
> > > 可以拿到，并且所有操作均可以使用。
> > 
> > 
> > 这个so文件怎么生成啊，使用上面的[ndk-pkg](https://github.com/leleliu008/ndk-pkg) 吗？ 这个ndk-pkg要求在linux环境才可以使用...
> 
> 过程很繁琐，而且需要改源代码。 你如果不是很精通Android与C++，最好不要弄。

android应用层开发，C++不行，难搞了

xanadu-lee, 2023-06-21:
> 

你如果是不会C/C++，就不用研究了，这玩意不是编译好就能用的。
具体的涉及一些事情，不能说。

ben1681, 2023-09-07:
@AiXanadu Hello, first of all thank you for your answer under [this question]( https://github.com/libusb/libusb/issues/717) to me. I read all the conversations here, The solution you mention here requires converting usbmuxd to a dynamic library, which is probably too difficult for me at this stage. I was at about the same stage of development as @hcropps at the time. I now have an immature idea, would like to ask whether it is feasible: 1. In the java code to obtain the usb device descriptor; 2. Modify the usbmuxd code so that it can accept the passed device descriptor as a parameter and use this parameter to initialize libusb; 3. Compile usbmuxd and place related files in the accessible path of the app. Start usbmuxd in java code and pass in the obtained device descriptor.
Can this method make usbmuxd work properly on devices without root theoretically? And if it is convenient, please give me some more guidance and tips, thank you very much!

> @hcropps Do as I said last time. Build all dependencies as lib(.a) libraries. Then link all libraries in your own dll(.so). Of course, you also need to build usbmuxd as a lib(exe >> lib) library. [...]

AiXanadu, 2023-09-07:
@ben1681 
First of all, the idea is very close, but you have to know that you are not allowed to store executable files in the Android package.
So you have to build usbmuxd as a dynamic library.

ben1681, 2023-09-07:
@AiXanadu I see in [this project](https://github.com/cczhr/OTGUninstallApps) that the author has put the executable under the assets directory, which gives me some inspiration. But then I will make further attempts according to your method. Thank you very much for replying to me so soon.

AiXanadu, 2023-09-07:
@ben1681 You can see if it requires Root.
As for me, I did this by compiling as a dynamic library.

ben1681, 2023-09-07:
@AiXanadu ok,I ll try it without Root.Can you give me a few more key steps on compiling as a dynamic library?

AiXanadu, 2023-09-07:
@ben1681 
certainly. . . . It's impossible. . .
Sorry, it's not that I don't want to help you. Hope you understand.

ben1681, 2023-09-07:
@AiXanadu I understand. Thank you very much anyway!

fgj54jrt, 2024-08-18:
> @hcropps libusb_wrap_sys_device 实际上并没有做任何有用的事情。 而且，文件描述符是 libusb 主动询问 Java，而不是 Java 主动调用 libusb。
> 
> 我的电子邮件地址是[libxanadu@gmail.com](mailto:libxanadu@gmail.com)

阿哈，没想到几年前做得项目，在这还能找到痕迹

fgj54jrt, 2024-08-18:
> @hcropps需要将 usbmuxd 转为静态库，然后将其与 libusb、libimobiledevice 等一起构建为动态库， 这样它们就共用一个 libusb。 然后手动在动态库中启用一个 usbmuxd 线程来运行该服务。
> 
> 有一个app也是这么做的， http://www.guomil.com/ 这里面的liblinux.so就是把依赖库打包成动态库，然后进行封装。

阿哈，没想到几年前搞的项目，在这还能找到痕迹

qq471077336, 2024-11-07:
> @ben1681 certainly. . . . It's impossible. . . Sorry, it's not that I don't want to help you. Hope you understand.

大佬，我这边有个关于usbmuxd的项目需要帮助，希望可以帮助我们，我们可以付出一些报酬，望回复，谢谢。

qq471077336, 2024-11-07:
@AiXanadu 
大佬，我这边有个关于usbmuxd的项目需要帮助，希望可以帮助我们，我们可以付出一些报酬，望回复，谢谢。