libimobiledevice/libimobiledevice #1064: Wich restore settings to use?

wuast94, 2020-11-08:
Hey all, didnt found a proper info about this and the man page is a bit confusing to me

i have made a backup and now want to test the restore, but wich flags to use to get all the data and settings back? 

Turned icloud off and encryption on (so wlan passwords and mail accounts are included i hope)

should i now just mak a restore? or do i need flags like:
"--system" restore system files, too. <-- too? are they not included when not include this flag? or are then only the system files restored? 
"--settings" same as system ,,, are then only settings restored or do i need to use this flag to restore the settings too
"--no-reboot" reboot the system when done. <-- no reboot flag to reboot the device when restore done? 
"--copy" create a copy of backup folder before restoring. <-- why should i make a copy from the backup i already have? or is the restore proccess changing files from my backup ? 
"--remove" remove items which are not being restored. <-- items that are NOT restored would be the last one i would delete .. but i think there are usecases for this, and i dont want to delte my backup anyway so this option i will never touch anyway 

sry when this questions are stupid but i am confused so much from this man page ...

wuast94, 2020-11-08:
in addition:

My actual phone will be repaired .. and i will get a new one so i need to restore it

nikias, 2020-11-08:
> Hey all, didnt found a proper info about this and the man page is a bit confusing to me
> 
> i have made a backup and now want to test the restore, but wich flags to use to get all the data and settings back?
> 
> Turned icloud off and encryption on (so wlan passwords and mail accounts are included i hope)
Yes, encryption will let you keep passwords etc.

```
idevicebackup2 --source OLD_UDID -i restore --system --settings
```