libimobiledevice/libimobiledevice #1066: idevicebackup2 hangs in poll_s

Ben-Voris, 2020-11-10:
On both an iPhone 6S Plus and an iPad Pro (10.5-inch), both running version 13.7, idevicebackup2 hangs. After the hang, `ps -l` shows WCHAN=poll_s

Before the backup hangs, /proc/<pid>io shows increasing values, except cancelled_write_bytes, which is zero. After the hang, cancelled_write_bytes is 12288. The values do not change after this, even after waiting > 10 hours.

```
rchar: 49049
wchar: 5116558311
syscr: 27
syscw: 355057
read_bytes: 139264
write_bytes: 5149847552
cancelled_write_bytes: 12288
```

The last few lines of output are:

```
Removing 1 file
[==================================================] 100% Finished
[==================================================] 100% Finished
Sending 'f4b5dfdcaa24454b049b9faee418159d715d471b/Status.plist' (189 Bytes)
Sending 'f4b5dfdcaa24454b049b9faee418159d715d471b/Manifest.plist' (101.4 KB)
```

The rotating icon on the devices continues to display for as long as I leave the device in this state.

The command is just `idevicebackup2 --debug backup --full ./ipad/`

Is there anything I can do to get these backups to complete?

Ben-Voris, 2020-11-11:
The provided syntax doesn't seem to work

```
: idevicesyslog -p BackupAgent2 -p backupd -p ReportCrash
Usage: idevicesyslog [OPTIONS]
Relay syslog of a connected device.

  -d, --debug		enable communication debugging
  -u, --udid UDID	target specific device by its 40-digit device UDID
  -h, --help		prints usage information

Homepage: <http://libimobiledevice.org>
```

If I run without any options, I get a lot of output. But I also get output when doing nothing.

nikias, 2020-11-11:
@Ben-Voris you are using an outdated version.

Ben-Voris, 2020-11-11:
I and failed to to build the current version. See [autogen fails with `No package 'libusbmuxd-2.0' found`](https://github.com/libimobiledevice/libimobiledevice/issues/1067).

Ben-Voris, 2020-11-11:
The last few syslog entries returned by `idevicesyslog | grep -E 'BackupAgent2|Crash|backupd'` are:

```
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Finished scanning domain <private> - size:26353662, files:525, dirs:82, clones:511, hardlinks:0
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Finished scanning All domains - total size:393335257, total files:3637, total dirs:3205, total clones:525, total hardlinks:6
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Scanned in 21.025 s
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Starting to prepare for preflight
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: uploadSize:47775792, uploadFileCount:481
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Free disk space on drive: 2578023206912
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Prepared for preflight in 0.001s
Nov 10 19:11:09 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: MBDriveBackupEngine, uploadBatchSize:128, concurrentUploadBatchCount:1
Nov 10 19:11:11 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Finished uploading 4 batches, s:{0,0,0,0}, t:1.551, r:0.000
Nov 10 19:11:12 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: ManifestDB: closing <private>
Nov 10 19:11:12 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: ManifestDB: opening <private>
Nov 10 19:11:13 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Uploaded 481 files of 47.7 MB in 3.189 s at 14.9 MB/s
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Moved 18133 files in 1.952 s
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Removed 0 items in 0.076 s
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: ManifestDB: closing <private>
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: ManifestDB: removing <private>
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Backup status: <private>
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Verifying backup contents
```

mexmer, 2020-11-11:
```
Nov 10 19:11:15 bvoriss-iPhone-6 BackupAgent2[264] <Notice>: Verifying backup contents
```
this operation actually tkaes a lot of time to finish, so it might look like it hangs, depending on backup size it might even be half hour or more. on my ipad air 3 (64GB) i have like 20GB backup and it takes around 17 minute.

Ben-Voris, 2020-11-11:
On the iPhone, Settings reports iPhone Storage of 54 GB of 64 GB used. du
reports its backup directory to be 15G. As mentioned, I left its backup in
this state for > 10 hours. Getting to the hang takes only a few minutes.

On the iPad, Settings reports iPad Storage reports 43 GB of 256 GB used.
The backup directory is 4.8G.. I left its backup in this state for a few
tens of minutes.

For comparison, a backup of an old iPhone, which I've wiped, so I don't
know how full it was, but which didn't hang has a backup directory of 1.7
G. This, using the same computer, cable, and USB port. Of course, I don't
know if it reported the canceled write.

When in this state, idevicebackup2 uses no CPU time and does no i-o (at
least so far as the /proc filesystem can be trusted). What *should* it be
doing? Are there any diagnostics that I can enable?

mexmer, 2020-11-13:
hours is too much definetly, should be minutes or tenth of minutes.

sounds like communication was somehow broken, so it now doesn't receive any data, but why it's deadlocked i can't tell.

i tried latest imobilebackup2 from source, and don't observe such issue either on linux or windows.

but since you posted in other threads, that you might have outdated versions, that of course might be also causing this issue.