libimobiledevice/libimobiledevice #1067: autogen fails with `No package 'libusbmuxd-2.0' found`

Ben-Voris, 2020-11-11:
On Ubuntu 18.04, autogen fails with `No package 'libusbmuxd-2.0' found`

## Install packages from [README](https://github.com/libimobiledevice/libimobiledevice/blob/master/README.md)

```
: sudo apt-get install \
> build-essential \
> checkinstall \
> git \
> autoconf \
> automake \
> libtool-bin \
> libplist-dev \
> libusbmuxd-dev \
> libssl-dev \
> usbmuxd

Reading package lists... Done
Building dependency tree       
Reading state information... Done
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.15.1-3ubuntu2).
build-essential is already the newest version (12.4ubuntu1).
libplist-dev is already the newest version (2.0.0-2ubuntu1).
libtool-bin is already the newest version (2.4.6-2).
libusbmuxd-dev is already the newest version (1.1.0~git20171206.c724e70f-0.1).
checkinstall is already the newest version (1.6.2-4ubuntu2).
git is already the newest version (1:2.17.1-1ubuntu0.7).
libssl-dev is already the newest version (1.1.1-1ubuntu2.1~18.04.6).
usbmuxd is already the newest version (1.1.0-2ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
```

## git clone

```
: git clone https://github.com/libimobiledevice/libimobiledevice.git && cd libimobiledevice
Cloning into 'libimobiledevice'...
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 9242 (delta 3), reused 8 (delta 3), pack-reused 9224
Receiving objects: 100% (9242/9242), 2.86 MiB | 18.32 MiB/s, done.
Resolving deltas: 100% (6537/6537), done.
```

## Failed autogen

```
: ./autogen.sh
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:29: installing './compile'
configure.ac:32: installing './config.guess'
configure.ac:32: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd... no
configure: error: Package requirements (libusbmuxd-2.0 >= 2.0.2) were not met:

No package 'libusbmuxd-2.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

mexmer, 2020-11-11:
well, it tells you that you missing recent version of libusbmuxd, so first download and compile libusbmuxd 

https://github.com/libimobiledevice/libusbmuxd

and don't forget to set PKG_CONFIG_PATH afterwards

Ben-Voris, 2020-11-11:
I'm new to desktop Linux, but I had thought that the apt-get output `usbmuxd is already the newest version (1.1.0-2ubuntu0.1)` told me that I have the latest. But when I clone https://github.com/libimobiledevice/libusbmuxd.git, I see that it includes tags up to 2.0.2 (and no 1.1* tag). With that, it is true to say that what apt-get is telling me that I have the latest that Ubuntu provides for their 18.04 release, which is not nearly the latest?

When I run autogen.sh for libusbmuxd, it fails with:

```
checking for libplist... no
configure: error: Package requirements (libplist-2.0 >= 2.2.0) were not met:

No package 'libplist-2.0' found
```

I found https://github.com/libimobiledevice/libplist.git and built it using its instructions, which allowed me to build libusbmuxd, which allowed me to build libimobiledevice.

I don't understand what PKG_CONFIG_PATH is and did not set it. [pkg-config(1)](https://linux.die.net/man/1/pkg-config), which says it only affects .pc files and I don't know they are. Please, would you give a specific example?

Also, I wonder if the builds worked by accident of how I stored the cloned directories. That is:

libimobiledevice/
> libimobiledevice/
> libplist/
> libusbmuxd/

Or, because I did not set PKG_CONFIG_PATH, do I have a build programs that use a mismatch of library versions?

Fwiw, the idevicebackup2 program this produced is much smaller (6373 bytes) than the one shipped with Ubuntu 18.04 (213744 bytes).

I did run `sudo make install` for all three.

Ben-Voris, 2020-11-12:
My observation about file size is is irrelevant.

The size difference of the Ubuntu 18.084 idevicebackup2 and the smaller file I looked at exists because the former is a program and the latter is a bash script (it does some magic to use run the newly created .lib/idevicebackup2 with a modified LD_LIBRARY_PATH).

My question about how to build remains.

nikias, 2020-11-12:
```
sudo apt install build-essential checkinstall git autoconf automake libtool-bin libssl-dev pkg-config cython
git clone https://github.com/libimobiledevice/libplist
git clone https://github.com/libimobiledevice/libusbmuxd
git clone https://github.com/libimobiledevice/libimobiledevice
git clone https://github.com/libimobiledevice/usbmuxd
cd libplist
./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libusbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libmobiledevice
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --enable-debug
make && sudo make install
cd ..
cd usbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --prefix=/usr --sysconfdir=/etc --localstatedir=/var --runstatedir=/run
make && sudo make install
sudo killall usbmuxd
```
Unplug device and plug it back in, then try `idevice_id -l`, `ideviceinfo`, etc.

mexmer, 2020-11-13:
@Ben-Voris precompiled packages in linux distribution are not always up to date with codebase on project, reason is simple, when you make stable distribution, you don't include experimental stuff.

also even for testing/unstable branch of distribution, there is always some package maintainer, and since that person oversees multiple packages, it takes time for them to get them up to date.

if you compile from source code (like in this case), you usualy end with compiling almost every dependent package from source code. as @nikias posted above.

Jambon1510, 2021-02-27:
On my side, had a similar issue then take the @nikias script with some modifications cause it was not working out of the box (cd libimobiledevice, install libusb-1.0-0-dev + downloads cleanup )  as below to make it work:

```
sudo apt install build-essential checkinstall git autoconf automake libtool-bin libssl-dev pkg-config cython libusb-1.0-0-dev
git clone https://github.com/libimobiledevice/libplist
git clone https://github.com/libimobiledevice/libusbmuxd
git clone https://github.com/libimobiledevice/libimobiledevice
git clone https://github.com/libimobiledevice/usbmuxd
cd libplist
./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libusbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libimobiledevice
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --enable-debug
make && sudo make install
cd ..
cd usbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --prefix=/usr --sysconfdir=/etc --localstatedir=/var --runstatedir=/run
make && sudo make install
sudo killall usbmuxd

rm -fr libplist/ libusbmuxd/ libimobiledevice/ usbmuxd/
```

nikias, 2021-02-27:
@Jambon1510 thanks for the note about the missing package and my typo, I updated my comment above.

liquidsunshine, 2021-03-24:
> On my side, had a similar issue (using Ubuntu 20.04) then take the @nikias script with some modifications cause it was not working out of the box (fix `cd libimobiledevice`, install `libusb-1.0-0-dev` install + downloads cleanup ) as below to make it work:
> 
> ```
> sudo apt install build-essential checkinstall git autoconf automake libtool-bin libssl-dev pkg-config cython libusb-1.0-0-dev
> git clone https://github.com/libimobiledevice/libplist
> git clone https://github.com/libimobiledevice/libusbmuxd
> git clone https://github.com/libimobiledevice/libimobiledevice
> git clone https://github.com/libimobiledevice/usbmuxd
> cd libplist
> ./autogen.sh
> make && sudo make install
> sudo ldconfig
> cd ..
> cd libusbmuxd
> PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
> make && sudo make install
> sudo ldconfig
> cd ..
> cd libimobiledevice
> PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --enable-debug
> make && sudo make install
> cd ..
> cd usbmuxd
> PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --prefix=/usr --sysconfdir=/etc --localstatedir=/var --runstatedir=/run
> make && sudo make install
> sudo killall usbmuxd
> 
> rm -fr libplist/ libusbmuxd/ libimobiledevice/ usbmuxd/
> ```

This did the trick for me!  I had to do `sudo ldconfig` once more at the end to get rid of this error:

```
idevicebackup2: error while loading shared libraries: libimobiledevice-1.0.so.6: cannot open shared object file: No such file or directory
```

Lowclouds, 2021-11-03:
The build script above is now (11/3/21 Ubuntu 20.04) failing when it can't find libimobiledevice-glue-1.0

--------
following build tools install and git clone of repos:

/mnt/data/dev/libmobiledevice$ cd libusbmuxd
/mnt/data/dev/libmobiledevice/libusbmuxd$ PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
libtoolize: putting auxiliary files in '.'.
[ omitted successful stuff ]
checking for libplist... yes
checking for limd_glue... no
configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:

No package 'libimobiledevice-glue-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables limd_glue_CFLAGS
and limd_glue_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
/mnt/data/dev/libmobiledevice/libusbmuxd$ make && sudo make install
make: *** No targets specified and no makefile found.  Stop.

--------------
trying to install libimobiledevice-glue-1.0 fails with

E: Unable to locate package libimobiledevice-glue-1.0

finally, trying to clone and build libmobiledevice-glue-1.0 fails:

$ git clone https://github.com/libimobiledevice/libimobiledevice-glue-1.0
Cloning into 'libimobiledevice-glue-1.0'...
Username for 'https://github.com': 
Password for 'https://github.com': 
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/libimobiledevice/libimobiledevice-glue-1.0/'

nikias, 2021-11-04:
The repository is here:
https://github.com/libimobiledevice/libimobiledevice-glue

nikias, 2021-11-04:
```bash
sudo apt install build-essential checkinstall git autoconf automake libtool-bin libssl-dev libusb-1.0-0-dev pkg-config cython 
git clone https://github.com/libimobiledevice/libplist
git clone https://github.com/libimobiledevice/libimobiledevice-glue
git clone https://github.com/libimobiledevice/libusbmuxd
git clone https://github.com/libimobiledevice/libimobiledevice
git clone https://github.com/libimobiledevice/usbmuxd
cd libplist
./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libimobiledevice-glue
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libusbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
make && sudo make install
sudo ldconfig
cd ..
cd libimobiledevice
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --enable-debug
make && sudo make install
cd ..
cd usbmuxd
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --prefix=/usr --sysconfdir=/etc --localstatedir=/var --runstatedir=/run
make && sudo make install
sudo killall usbmuxd
```

Lowclouds, 2021-11-13:
Thanks. While I have many other questions about using these tools, I now have working versions.