libimobiledevice/libimobiledevice #1071: DLMessagePurgeDiskSpace new message is not processed in devicebackup2.c

bricedenice59, 2020-11-18:
Hi Guys, in devicebackup2.c I cannot see that new message "DLMessagePurgeDiskSpace" processed.
it is probably something new that has been introduced with iOS 14 as in the past it was sending back an error code = -105 in case of not enough free space to backup a device.
Could you add a fix for it ?

Thanks a million.

mexmer, 2020-11-18:
is that message sent by phone or by computer?

if it's sent by phone, question is, what exactly it does?
it might be just informational, since OS itself should not decide, that it will delete your files, because it needs space for device backup, or it might have UI interaction in OS X 10.15 or newer (where they moved iOS device from itunes to finder)

question also is, what is condition to trigger this message, is it appearing only if you don't have enough free space on disc, or can it appear also in different condition?
i have not seen this message yet, so we need to a way to trigger it.

before space required was part of MBError message, so it was only informational.

nikias, 2020-11-18:
It's coming from the device side. I will implement this as a no-op though, since I can't make the idevicebackup2 handle any disk purging (like empty trash etc.). I will try to see how the host side (macOS) is handling it.

bricedenice59, 2020-11-19:
Thanks @nikias , I confirm this comes from the phone and it is purely informational since that's the user's business to get the required free space to backup its device.

nikias, 2020-11-19:
I can't find any trace of this being handled on the host side (macOS). On the device side I can see there's an error message basically saying that the host doesn't understand this message. I guess that's what I have to send back to the device.

nikias, 2020-11-19:
Even worse, the DeviceLink framework actually disconnect when it receives an unknown message (where the above is considered unknown on macOS Catalina). Not sure what is going on here...

nikias, 2020-11-19:
If you can trigger this condition, you can try this patch:
```
diff --git a/tools/idevicebackup2.c b/tools/idevicebackup2.c
index 6412fe2..068420d 100644
--- a/tools/idevicebackup2.c
+++ b/tools/idevicebackup2.c
@@ -2252,6 +2252,11 @@ checkpoint:
                                        plist_t freespace_item = plist_new_uint(freespace);
                                        mobilebackup2_send_status_response(mobilebackup2, res, NULL, freespace_item);
                                        plist_free(freespace_item);
+                               } else if (!strcmp(dlmsg, "DLMessagePurgeDiskSpace")) {
+                                       /* device wants to purge disk space on the host - not supported */
+                                       plist_t empty_dict = plist_new_dict();
+                                       err = mobilebackup2_send_status_response(mobilebackup2, -1, "Operation not supported", empty_dict);
+                                       plist_free(empty_dict);
                                } else if (!strcmp(dlmsg, "DLContentsOfDirectory")) {
                                        /* list directory contents */
                                        mb2_handle_list_directory(mobilebackup2, message, backup_directory);
```
It will return an error code to the device, so the device will then abort the backup. Take a look at idevicesyslog and see if it prints anything interesting.

mexmer, 2020-11-19:
ok, i did simple test, just on request for disk space query i returned 1MB, then i got purge request, when i sent error response i got back

Insufficient free disk space on drive to back up (MBErrorDomain/105)

this time without required space number.

mexmer, 2020-11-19:
looks number after message is bytes required for backup, for differential backup it asked 2.5GB, for full backup 10GB, my current backup is roughly 7.8GB

no idea what other number means, tho.

bricedenice59, 2020-11-25:
Thanks for the fix @nikias

dinochen1020, 2022-06-30:
Does anyone knows the meaning of second number in dlmessage now?

nikias, 2022-06-30:
@mexmer you don't have by any chance a full sample message plist?

mexmer, 2022-06-30:
@nikias 
i did cut begining of communication, but here is communication since free space request ... just tested it against iOS 15.6
> phone
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <array>
    <string>DLMessageGetFreeDiskSpace</string>
  </array>
</plist>
```
> computer
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <array>
    <string>DLMessageStatusResponse</string>
    <integer>0</integer>
    <string>___EmptyParameterString___</string>
    <integer>10000000</integer>
  </array>
</plist>
```
> phone
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <array>
    <string>DLMessagePurgeDiskSpace</string>
    <integer>26418026591</integer>
    <integer>4</integer>
  </array>
</plist>
```
> computer
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <array>
    <string>DLMessageStatusResponse</string>
    <integer>-1</integer>
    <string>Operation not supported</string>
    <string>___EmptyParameterString___</string>
  </array>
</plist>
```
> phone
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN"	"http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <array>
    <string>DLMessageProcessMessage</string>
    <dict>
      <key>ErrorCode</key>
      <integer>105</integer>
      <key>ErrorDescription</key>
      <string>Insufficient free disk space on drive to back up (MBErrorDomain/105)</string>
      <key>MessageName</key>
      <string>Response</string>
    </dict>
  </array>
</plist>
```

mexmer, 2022-11-29:
@nikias i finally found, what purge command does ... it works only for OSX apfs, if you notice on mac, when you check disk space, it says there is some available, used, total, and there is XY megabytes of purgeable space (it usually means clearing caches, temporary files, recycle bin, old versions of files). purge effectively call system function, that does clear this stuff, that operating systems thinks it's safe to clear, no idea how to do that programatically, but itunes can do that.

[image]