libimobiledevice/libimobiledevice #1073: Check if passcode is set

marchuknikolay, 2020-11-23:
Hello!
Could you say, please, is there a way to check if passcode is set now?

I've tried 2 methods:
1.  Get list of device properties via `lockdownd_get_value(lockdown, nullptr, nullptr, &node)` and get value of PasswordProtected property, but when I unlocked screen I received false, however passcode was still set.
2. Using diagnostics_relay_client_t:
```
diagnostics_relay_client_t diag = NULL;
plist_array_append_item(keys, plist_new_string("PasswordConfigured"));
diagnostics_relay_query_mobilegestalt(diag, keys, &dict)
node = plist_access_path(dict, 2, "MobileGestalt", "PasswordConfigured");
plist_get_bool_val(node, &passcode_hint)
```
However, it returns true for both my devices ios 14.1 with passcode, ios 12.4.8 without passcode

closed 2020-11-30 by marchuknikolay

hiepnh0, 2021-05-10:
@marchuknikolay 
You closed this issue, could you share your method to detect your device passcode is set or not?
Thanks.

sabadoryo, 2024-09-27:
@marchuknikolay @hiepnh0 still unanswered for now, were you able to solve it?