libimobiledevice/libimobiledevice #1080: pkg check modules

psxpetey, 2020-12-08:
libusbmuxd, libusbmuxd-2.0>= libusbmuxd version. Also I cant make the libimobile ( no targets specified no makefile found stop)

psxpetey, 2020-12-08:
upgraded libplist to 2.0 and same damn problem the instructions seem outdated as well a there is no autogen in the updated files why is this happening

nikias, 2020-12-09:
This is not how to report issues. What instructions are outdated? It works fine if you build everything in order. Also next time please use the search function, this has been answered too many times already. See my comment here:
https://github.com/libimobiledevice/libimobiledevice/issues/1067#issuecomment-726234592

closed 2020-12-09 by nikias

psxpetey, 2020-12-09:
> This is not how to report issues. What instructions are outdated? It works fine if you build everything in order. Also next time please use the search function, this has been answered too many times already. See my comment here:
> [#1067 (comment)](https://github.com/libimobiledevice/libimobiledevice/issues/1067#issuecomment-726234592)

I did everything in order even installed different dists of linux. Why does it demand libusbmux 2.0 when I pulled it just like it says in the instructions, so then I have to install libplist 2.0 then it allows libusbmux and then libimobile just fails and asks for 2.0 when its clearly installed manually and non manually. maybe ldconfig for every single step?

someone on another forum using arch said they installed as per instructions no issues. Ive used santoku mint and unbuntu (santoku is lunbuntu).