libimobiledevice/libimobiledevice #1082: libimobile wont build says ./configure: line 16089: `PKG_CHECK_MODULES(libusbmuxd, libusbmuxd-2.0 >= $LIBUSBMUXD_VERSION)'

psxpetey, 2020-12-09:
I installed all as per instructions in exact order, I then try to build libimobile I get this error: (I used git clone and then tried 1.3 source and now 1.2 source on 1.3 or regular clone I get libusbmuxd2.0

checking how to hardcode library paths into programs... immediate
./configure: line 16089: syntax error near unexpected token `libusbmuxd,'
./configure: line 16089: `PKG_CHECK_MODULES(libusbmuxd, libusbmuxd-2.0 >= $LIBUSBMUXD_VERSION)'

./configure: line 16034: syntax error near unexpected token `libusbmuxd,'
./configure: line 16034: `PKG_CHECK_MODULES(libusbmuxd, libusbmuxd >= $LIBUSBMUXD_VERSION)'
s@s-VirtualBox:~/libimobiledevice-1.2.0$ 
I then build libmux 2 or above and get "need libplist " I upgrade libplist" builds and installs

libimobile wont build is there a way to fix why does itkeep saying 2.0 when i followed the instructions to a t

nikias, 2020-12-09:
Why do you open a new issue? Didn't you ask about this in #1080?
If you want help, state where you installed libplist and libusbmuxd. Also:
Do you have pkg-config installed?
Did you set PKG_CONFIG_PATH?

psxpetey, 2020-12-09:
> Why do you open a new issue? Didn't you ask about this in #1080?
> If you want help, state where you installed libplist and libusbmuxd. Also:
> Do you have pkg-config installed?
> Did you set PKG_CONFIG_PATH?

Here's a better question where is libusbmux supposed to be installed I could set the config path for that and then id be hunky dorey.  No matter how much I search it wont tell me where libusbmux is intalled even with dpkg it will tell me that it is installed and the version but not where.  libplist is in /usr/local/lib there is a libmux.la so and so 6 there should it be in /sbin?

nikias, 2020-12-09:
If you don't specify anything to `./autogen.sh` or `./configure` it should usually end up in `/usr/local` prefix (though the default may vary with the distribution), so headers will be in `/usr/local/include` and libs in `/usr/local/lib/libusbmuxd.*`. pkg-config files will end up in `/usr/local/lib/pkgconfig/*.pc`. The thing is, if you installed libusbmuxd after build using `sudo make install` it will actually print `libraries have been installed in ...`.
Anyway did you actually try what I mentioned here: https://github.com/libimobiledevice/libimobiledevice/issues/1067#issuecomment-726234592 ?

psxpetey, 2020-12-09:
> If you don't specify anything to `./autogen.sh` or `./configure` it should usually end up in `/usr/local` prefix (though the default may vary with the distribution), so headers will be in `/usr/local/include` and libs in `/usr/local/lib/libusbmuxd.*`. pkg-config files will end up in `/usr/local/lib/pkgconfig/*.pc`. The thing is, if you installed libusbmuxd after build using `sudo make install` it will actually print `libraries have been installed in ...`.
> Anyway did you actually try what I mentioned here: [#1067 (comment)](https://github.com/libimobiledevice/libimobiledevice/issues/1067#issuecomment-726234592) ?

when installed libusbmuxd-2.0.1 says it installed libraries in   /usr/local/lib 
if I do pkg_config to that path it says No package 'libusbmuxd-2.0' found
how can I get libimobile to see libusbmuxd

nikias, 2020-12-09:
You need to specify the pkgconfig dir, not the libdir. Try:
```
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig pkg-config --libs libusbmuxd-2.0
```
If that prints
```
-lusbmuxd-2.0
```
Then all is good and you can follow my steps in the link above to my comment in the other ticket.

psxpetey, 2020-12-09:
thanks works

closed 2020-12-09 by psxpetey