libimobiledevice/libimobiledevice #1084: [Data loss] idevicebackup2 does not verify password when enabling encryption

metronidazole, 2020-12-10:
Unfortunately idevicebackup2 doesn't provide any way to verify a chosen encryption passphrase when enabling encryption.

The issue with this is that if you make a typo (or multiple), all your backups are 100% useless unless you can guess what the actual passphrase used was. I believe the official iTunes client does force you to enter your passphrase twice to prevent this scenario.

nikias, 2020-12-13:
Hi, I am not sure if I am missing something here, but it does ask for you to re-type the password:
```
❯ idevicebackup2 -i encryption on
Started "com.apple.mobilebackup2" service on port 50338.
Negotiated Protocol Version 2.1
Enter new backup password: **********
Enter new backup password (repeat): **********
Backup encryption has been enabled successfully.
```

metronidazole, 2020-12-14:
Strange, will retest and let you know if it happens again.

closed 2020-12-14 by metronidazole