libimobiledevice/libimobiledevice #1086: Segmentation Fault on backup

benwar89, 2020-12-11:
Hi, i got a Segmentation fault when i want to do a backup of my  phone.
i build libimobiledevice from source on my raspberry zero W running raspberry os lite
my phone is the iPhone 12 Pro running the last version of IOS 14

here the output of the consol 

Backup directory is "/home/pi/ios_backup/"
Started "com.apple.mobilebackup2" service on port 49883.
Negotiated Protocol Version 2.1
Starting backup...
Backup will be encrypted.
Requesting backup from device...
Full backup mode.
Segmentation fault

nikias, 2020-12-13:
Any chance you can run it in a debugger? You might have to install it first: `sudo apt install gdb`
Then run:
```
gdb --args idevicebackup2 backup yourbackupdir
```
Then in the `gdb` shell run:
```
(gdb) r
```

And wait until it signals with the segfault, then type
```
(gdb) bt
```
And paste that output here.

cyberodoma, 2021-03-24:
Hi, I saw this and was wondering if this ever got resolved. I'm currently experiencing a similar issue. Following the above comments here's my output for triage: 
[image]

benwar89, 2021-03-24:
Hi, i tried to debug the program, the error happen when the program try to create the folder who will contain the backup. Also this problem only happen when the destination folder are in a removable disk ( a USB key for me). I didn’t go to the bottom of that but you can create the folder yourself before launching the backup or use don’t use a removable disk 
I hope this can help you :)

cyberodoma, 2021-03-24:
Thanks @benwar89! Your suggestion worked for me. I created my backup folder in the /opt directory. moving this to a folder on my desktop resolved it. Thanks once again!

seohissa, 2022-08-27:
[image]
@nikias 大神！我也遇到了同样的问题，没有思路。能不能指点一下，谢谢

taozuia, 2022-10-02:
> [image]
> 
> @nikias 大神！我也遇到了同样的问题，没有思路。能不能指点一下，谢谢

时隔一个月再次遇到这个问题。是配对没配好

datatags, 2023-04-08:
> you can create the folder yourself before launching the backup

This was it for me, I tried creating the folder manually and discovered there was a permissions issue that was preventing it from creating the folder. `sudo chown myuser:myuser /path/to/backup/folder` resolved it, and I didn't even need to create the folder manually after fixing the permissions! 😄