libimobiledevice/libimobiledevice #1088: Boot loop on restore

Linkandzelda, 2020-12-13:
iPhone 6S iOS 14.2, did a backup which was successful. Did a restore which was successful, but it boot loops with --settings option.

demhademha, 2020-12-13:
So it bootloops when you go to the settings app?

nikias, 2020-12-13:
@Linkandzelda well the thing is even with iTunes the restore would have resulted in the same state, if the backup is to blame here. idevicebackup2 does the same as iTunes (if you pass --system --settings options). Did you have the device jailbroken? Maybe some filesystem modifications resulted in the bootloop state. Also, is it really bootlooping, like does the screen really go completely dark and boots again? Or maybe is it just Springboard restarting?

@demhademha no, --settings is a restore option that you can pass to `idevicebackup2 restore`. This option is also enabled when you restore with iTunes.

codegoddesskh, 2020-12-13:
Try factory restoring/resetting the iPhone ("Restore iPhone" in iTunes) and restoring the backup again to the blank, not set up iPhone. (Just like you did before, no need to change the restore method.) I've seen an issue a lot since iOS 14.2 where restoring a backup to a set up iPhone causes the boot loops after the restore. Usually everything works fine when the same backup is restored to the blank iPhone instead.

Linkandzelda, 2020-12-14:
To add further to this, the device was not jailbroken and I used --system and --settings. I did a restore with iTunes to fix the loop but the first time this failed. The 2nd time the restore worked without issues. I then restored the backup without --settings and without --system, and this worked as expected.

nikias, 2020-12-14:
Oh interesting. Maybe Apple disabled the system restore setting. I will check.