libimobiledevice/libimobiledevice #1089: About data synchronization of equipment

XanaduAPI, 2020-12-16:
Can I use libimobile device to synchronize music from my device? Mobilesync doesn't seem to be the feature I'm looking for.

tuanjnfor, 2020-12-18:
Dear @XanaduAPI ,
Sorry, it is not answer for your question.
But I'm building this lib for android application.
I see your guide in [here](https://github.com/libimobiledevice/libimobiledevice/issues/1061) and [here](https://github.com/libimobiledevice/usbmuxd/issues/53) 
But I can't find ARM_64 folder in ANDROID_HOME path
ANDROID_ARM_64=$ANDROID_HOME/ARM_64
I'm using Ubuntu 20.04
Pls help to let me know how can I find ARM_64 for building lib.
Thank you so much!

XanaduAPI, 2020-12-19:
@tuanjnfor 
NDK_HOME=/home/xanadu/Tools/android-ndk-r21b
JDK_HOME=/home/xanadu/Tools/jdk1.8.0_271
ANDROID_HOME=/home/xanadu/Android
BUILD_HOME=/home/xanadu/Build

export PATH=$NDK_HOME:$PATH
export PATH=$JDK_HOME:$PATH
export PATH=$ANDROID_HOME:$PATH
export PATH=$BUILD_HOME:$PATH

ANDROID_ARM_32=$ANDROID_HOME/ARM_32
ANDROID_ARM_64=$ANDROID_HOME/ARM_64
ANDROID_X86_32=$ANDROID_HOME/X86_32
ANDROID_X86_64=$ANDROID_HOME/X86_64

mexmer, 2020-12-20:
@XanaduAPI libimobile has no such functionality. but you can use libimobile + ifuse to mount filesystem from you ios device, there is folder with itunes synced media which will be accessible in R/W mode trough ifuse.

still you need to figure out yourself what to modify, and where to place data, to successfully sync them with computer.

i've seen some commerecial software for iOS media synchronization, so it's possible, but it's out of scope of this project functionality.

XanaduAPI, 2020-12-21:
@mexmer  Can it simulate ATH.exe sync?

tuanjnfor, 2020-12-21:
Dear @XanaduAPI,
Thank you for your help!
I can config to build.

I'm porting lib usbmuxd for android.
As your guide in [here](https://github.com/libimobiledevice/usbmuxd/issues/53) 
[image]

I understood that, I have to build 
libplist -> libusbmuxd -> openssl -> libimobiledevice-> libusb
And final is usbmuxd

But I can't find project openssl to build as below command
openssl
./Configure --cross-compile-prefix=arm-linux-androideabi- --prefix=$ANDROID_ARM_32 linux-armv4 no-asm no-engine no-ui

Where can I find opessl to  build with it?
I know it is not  topic for this. I'm sorry about that!
My email: tuanjnfor@gmail.com
Can you help to share your email to me?
I'm having task to connect android to iphone device by usb cable.
It is better for me if I can have your help by email or other communication.
Thank you so much!

XanaduAPI, 2020-12-21:
@tuanjnfor 
I think you can download OpenSSL here https://www.openssl.org/source/ 
My email:8059826@qq.com

XanaduAPI, 2020-12-21:
@tuanjnfor 
My mail was marked and returned

closed 2020-12-21 by XanaduAPI

hcropps, 2022-10-03:
@tuanjnfor can you solve the problem?