libimobiledevice/libimobiledevice #1095: Is there a command to get battery health?

c0ry, 2020-12-23:
In an iPhone's Battery settings there is "Battery Health" Maximum Capacity, which will show how degraded the battery is.  Is there a command I could run to get this percentage?

I tried idevicediagnostics and that shows FullChargeCapacity, but this is always 100.

Does anyone know where I could get this information via a USB connection without installing an app on the phone?

a-gunter, 2020-12-23:
IOReg has this information under different entries depending on the model.

"idevicediagnostics ioregentry AppleARMPMUCharger" for older models (iPhone 7 and older)
"idevicediagnostics ioregentry AppleSmartBattery" for newer models (iPhone 8 and newer)

You can find information about the battery, including design capacity & current capacity

c0ry, 2020-12-29:
Thanks, this is showing me a lot more information.   Is there documentation somewhere that I can find all the entry names available?  Do you know if there is something that could give me the LCD serial number?

itsferro, 2025-02-15:
does anyone have an idea on how I can get theis commands results with python on windows?
"idevicediagnostics ioregentry AppleARMPMUCharger" for older models (iPhone 7 and older)
"idevicediagnostics ioregentry AppleSmartBattery" for newer models (iPhone 8 and newer)