libimobiledevice/libimobiledevice #1104: Could not start com.apple.debugserver!

janksenhu, 2020-12-26:
idevicedebug -u 00008101-000B65DE2191003A run "cc.hudsonbld.TestProj" "argument='127.0.0.1'" "arguments2=9090"
Could not start com.apple.debugserver!
Please make sure to mount the developer disk image first:
  1) Get the iOS version from `ideviceinfo -k ProductVersion`.
  2) Find the matching iPhoneOS DeveloperDiskImage.dmg files.
  3) Run `ideviceimagemounter` with the above path.

jpstotz, 2021-01-11:
Make sure you are using the latest version of libimobiledevice:

```
brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies ideviceinstaller
brew uninstall --ignore-dependencies usbmuxd
brew uninstall --ignore-dependencies libplist
sudo rm /var/db/lockdown/*
brew install --HEAD libplist
brew install --HEAD usbmuxd
brew unlink usbmuxd
brew link usbmuxd
brew install --HEAD libimobiledevice
brew install --HEAD ideviceinstaller
```

denty, 2021-01-15:
My DeviceInfo:
Xcode 12.3
iOS 14.0
iphone XR
@janksenhu 
@jpstotz using the latest version of libimobiledevice will resolve this issue.