libimobiledevice/libimobiledevice #1105: idevicebackup2 does not recognize my password

Dadoum, 2020-12-29:
I first backed up my phone on iOS 13.3 with the `libimobiledevice` package on Arch Linux. The backup was encrypted and I wrote a password with `-p` command line arg.

Then, I upgraded to iOS 14.1, and wanted to restore the backup. It seems that the Arch Linux's package `libimobiledevice` is outdated so I installed `libimobiledevice-git to get a more recent version But now, the backup won't restore and it complains of an incorrect password. I tried with both `-i` and `-p`. I tried to move the backup to a Windows computer and restore it via iTunes (password incorrect). And I am sure that the password is correct since I written it as a command line arg and that my shell has enough history to see the command.

It could also be a device issue

nikias, 2020-12-29:
Hi, please understand that the backup password can only be verified by a device. If it says it is incorrect then it is just incorrect.
Check your script and verify that it didn't add a space or newline character or similar when it passed the password to idevicebackup2. Have you tried to modify the same script to trigger a restore?

Dadoum, 2020-12-29:
Yes, i grief

Dadoum, 2020-12-29:
Actually, the problem since to be way deeper: I tried to `unback`; my phone recognize the code and close the modal when I typed the correct password, but idevicebackup2 still cannot unback.

mexmer, 2020-12-29:
unback is broken since ios 10 or 11, it was mentioned here already.

unback command is processed by device, not libimobile. apple broke it and never fixed.

Dadoum, 2020-12-29:
Will try to downgrade to see if it is my phone version the problem; I have the exact bash history.

mexmer, 2020-12-29:
instead of downgrading ios, try to find out your password. might be just different letter case, or mistaken l (lower l) for 1 or I (upper i).

there are softwares out there, that can open encrypted ios backup if you know password, so you can use them to test password variants. 

when you find out correct one, just restore phone with idevicebackup2

Dadoum, 2020-12-29:
I know the exact password (sure at 100 % since I have the exact command I typed for backup, with `--password` flag). I copied the exact command on the exact same shell; the two only things that changed were idevicebackup2 (upgraded to latest one) and iOS version (13.3 -> 14.1)

mexmer, 2020-12-29:
@Dadoum 
backup command doesn't have --password parameter, it has only parameter for full backup, no other parameters, see
```
idevicebackup -h
```

password need to be set before by enabling encryption
```
idevicebackup2 encryption on 
```
so if you have not set password with encrypion command, then your backup is most likely unencrypted.
you can check if it's unecrypted by reading header of manifest.db file, if it's sqlite header, then your backup is unencrypted

mexmer, 2020-12-29:
also ios version doesn't matter for restore, you can always restore backup from older version to newer ios, i've done it many times

Dadoum, 2020-12-29:
Why it does this:

I type: `idevicebackup2 -p XXXXXX encryption on` -> you need -i to provide password

I type: `idevicebackup2 -p XXXXXX encryption on -i` -> Works, no prompt

I type: `idevicebackup2 encryption off -p something_else` -> Works ??

Dadoum, 2020-12-29:
Okay the issue was in the second command. the password was "-i"

closed 2020-12-29 by Dadoum

Matoran, 2021-07-23:
same error with `idevicebackup2 backup encryption on --password` thanks @Dadoum, password is `--password` :D

nikias, 2021-07-23:
The `-p` / `--password` option is to supply a password for the `restore` operation. For encryption on/off it is
```
encryption on|off [PWD]
```
If it shall be interactive, the `-i` needs to go in front:
```
idevicebackup2 -i encryption on
```
 I see that the options are a bit confusing and maybe I rewrite the option handling a bit.