libimobiledevice/libimobiledevice #1106: idevicebackup2: output not well suited for redirection into a file

ghollf, 2020-12-30:
The output of idevicebackup2 consists mostly of progress bars where each line is overwritten multiple times using carriage return. Nice for output on a terminal, but not so nice (and huge) redirected in a file.

It may be possible to achieve similar information without CR (e.g. when swapping information - first size and progressing hashes at the end).

nikias, 2020-12-31:
I will have it detect if stdout is a terminal and only then print a progress bar.

mexmer, 2020-12-31:
maybe add it as optional parameter instead.

ghollf, 2021-10-03:
Output looks like this (on an empty backup)...
```
[========================                          ]  48% Finished
Receiving files

[=                                                 ]   0% (42 Bytes/38.3 MB)     
[=                                                 ]   0% (785 Bytes/38.3 MB)     
[=                                                 ]   0% (43.2 KB/38.3 MB)     
[=                                                 ]   0% (83.2 KB/38.3 MB)     
[=                                                 ]   1% (345.3 KB/38.3 MB)     
[=                                                 ]   2% (607.5 KB/38.3 MB)     
[==                                                ]   2% (869.6 KB/38.3 MB)     
[==                                                ]   3% (1.1 MB/38.3 MB)     
[==                                                ]   4% (1.4 MB/38.3 MB)     
[===                                               ]   4% (1.7 MB/38.3 MB)     
[===                                               ]   5% (1.9 MB/38.3 MB)     
[===                                               ]   5% (2.1 MB/38.3 MB)     
[===                                               ]   6% (2.1 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.4 MB/38.3 MB)     
[====                                              ]   6% (2.5 MB/38.3 MB)     
[====                                              ]   6% (2.5 MB/38.3 MB)     
[====                                              ]   7% (2.5 MB/38.3 MB)     
[====                                              ]   7% (2.5 MB/38.3 MB)     
[====                                              ]   7% (2.5 MB/38.3 MB)     
[====                                              ]   7% (2.8 MB/38.3 MB)     
[=====                                             ]   8% (3.1 MB/38.3 MB)     
[=====                                             ]   9% (3.3 MB/38.3 MB)     
[=====                                             ]   9% (3.6 MB/38.3 MB)     
[======                                            ]  10% (3.9 MB/38.3 MB)     
[======                                            ]  11% (4.1 MB/38.3 MB)     
[======                                            ]  11% (4.4 MB/38.3 MB)     
[=======                                           ]  12% (4.6 MB/38.3 MB)     
[=======                                           ]  13% (4.9 MB/38.3 MB)     
[=======                                           ]  13% (5.2 MB/38.3 MB)     
[=======                                           ]  14% (5.3 MB/38.3 MB)     
[=======                                           ]  14% (5.3 MB/38.3 MB)     
[=======                                           ]  14% (5.3 MB/38.3 MB)     
[=======                                           ]  14% (5.3 MB/38.3 MB)     
[========                                          ]  14% (5.5 MB/38.3 MB)     
[========                                          ]  15% (5.8 MB/38.3 MB)     
[========                                          ]  16% (6.1 MB/38.3 MB)     
[=========                                         ]  16% (6.3 MB/38.3 MB)     
[=========                                         ]  16% (6.3 MB/38.3 MB)     
[=========                                         ]  16% (6.3 MB/38.3 MB)     
[=========                                         ]  17% (6.4 MB/38.3 MB)     
[=========                                         ]  17% (6.4 MB/38.3 MB)     
[=========                                         ]  17% (6.5 MB/38.3 MB)     
[=========                                         ]  17% (6.5 MB/38.3 MB)     
[=========                                         ]  18% (6.8 MB/38.3 MB)     
[==========                                        ]  18% (7.1 MB/38.3 MB)     
[==========                                        ]  19% (7.3 MB/38.3 MB)     
[==========                                        ]  20% (7.6 MB/38.3 MB)     
[===========                                       ]  20% (7.8 MB/38.3 MB)     
[===========                                       ]  21% (8.1 MB/38.3 MB)     
[===========                                       ]  22% (8.4 MB/38.3 MB)     
[============                                      ]  23% (8.6 MB/38.3 MB)     
[============                                      ]  23% (8.9 MB/38.3 MB)     
[============                                      ]  24% (9.1 MB/38.3 MB)     
[=============                                     ]  25% (9.4 MB/38.3 MB)     
[=============                                     ]  25% (9.7 MB/38.3 MB)     
[=============                                     ]  26% (9.9 MB/38.3 MB)     
[==============                                    ]  27% (10.2 MB/38.3 MB)     
[==============                                    ]  27% (10.5 MB/38.3 MB)     
[===============                                   ]  28% (10.7 MB/38.3 MB)     
[... 409 lines cut ...]

[================================================= ]  97% Finished
Moving 128 files

[... cut ...]
[==================================================]  99% Finished
Moving 1 file

[==================================================]  99% Finished
Moving 1 file

[==================================================] 100% Finished

[==================================================] 100% Finished
Removing 1 file

[==================================================] 100% Finished
Removing 1 file

[==================================================] 100% Finished

[==================================================] 100% Finished
```

ghollf, 2021-10-03:
So would still be great if progress output could be improved to be log file friendly.