libimobiledevice/libimobiledevice #1109: where do I find apple application support

SarKaa, 2021-01-11:
So I saw that this needs usbmuxd, or apple application support for windows. I want to use this without installing itunes. In the itunes setup exe, I found a .msi installer called apple mobile device support 64. Is that the same one, because I seem to not have one called apple application support.

In short, is apple mobile device support the same thing as apple application support, just updated?

Thank you

mexmer, 2021-01-11:
Apple Application support is different package, that you need for itunes and also update to work.

but apple mobile device support is what you need to communicate with apple devices on windows.

SarKaa, 2021-01-12:
ok thanks

closed 2021-01-12 by SarKaa