libimobiledevice/libimobiledevice #1111: idevicebackup2: Could not perform backup protocol version exchange, error code -1

Moertel113, 2021-01-13:
I can't backup my device on Linux

`$ idevicebackup2 -d backup Backups 
Backup directory is "Backups"
Started "com.apple.mobilebackup2" service on port 55205.
Could not perform backup protocol version exchange, error code -1`

I couldn't find any recent similar issues. This happens on two of my systems. They run the newest version.

`$idevicebackup2 -v
idevicebackup2 1.3.0
`
iPhone 8 - iOS 14.0 and iOS14.3

Backing up the iDevice was working fine a few months ago. Is anyone having a similar issue?

mexmer, 2021-01-13:
protocol version exchange error suggest you not using latest version. you need to use version from source on github.

Moertel113, 2021-01-13:
Thanks, that was the problem. I build it from source and it is working fine now!

closed 2021-01-13 by Moertel113

DJManas, 2021-04-26:
iOS 14.4.2, iPhone 7, idevicebackup 1.3.1 (cloned today 26.4.2021), problem persist. Is there something I can do to analyze whats causing the issue?

mexmer, 2021-04-26:
make sure you don't have distibution binaries and dev package installed before you compile and run from source

DJManas, 2021-04-26:
Sorry that was it, Gentoo unmerged everything but 2 libraries. Had to uninstall usbmuxd and then reemerge both from git. Thanks.

cedricloneux, 2021-07-22:
Having the same error, version on homebrew isn't up to date as it's 1.3.0.