libimobiledevice/libimobiledevice #1112: Latest revision idevicebackup2 fails with error code -1

DCKcode, 2021-01-17:
When I run `idevicebackup2` against and iPhone 12 running iOS 14.3 using Fedora 33, I seem to run into compatibility issues:

```
$ tools/idevicebackup2 backup /some/directory/
Backup directory is "/some/directory"
Started "com.apple.mobilebackup2" service on port <port>.
Could not perform backup protocol version exchange, error code -1
```
I've seen this error reported previously in issues where people are running an old version of `idevicebackup2`. But that's not what's happening here for me as I built it from HEAD:

```
$ tools/idevicebackup2 -v
idevicebackup2 1.3.1
$ git reflog
09f2332 (HEAD -> master, origin/master, origin/HEAD) HEAD@{0}: clone: from https://github.com/libimobiledevice/libimobiledevice.git
```
Also running `usbmuxd` from git HEAD. Really appreciate you guys making this open source library - if there's anything I can do to figure out what's going on here let me know.

DCKcode, 2021-01-18:
I tested the same setup with an iPhone 6 running iOS 12.4.6 and it worked out of the box, hence my suspicion that there's an incompatibility in a recent iOS version.

nikias, 2021-01-18:
@DCKcode are you sure you configured with `--enable-debug` ? It should print some plist data in the debug output.

DCKcode, 2021-01-18:
How can I make sure? I've just reran it again:

```
$ make clean
$ ./autogen.sh --enable-debug
$ make
```
And still no luck:
```
$ tools/idevicebackup2 -d backup /some/directory
Backup directory is "/some/directory"
Started "com.apple.mobilebackup2" service on port <port>.
Could not perform backup protocol version exchange, error code -1
```

mexmer, 2021-01-18:
you have most likely another version of libimobile library already installed in system, and tool is using that library, instead of your manually built libimobile.

make sure to uninstall fedora libimobile package, before compiling and installing your own build from github source.

nikias, 2021-01-18:
@mexmer I am suspecting the same, but _usually_ running from the build tree uses the in-tree built libs. This could be different in the above case.

mexmer, 2021-01-18:
@nikias he is running it from tools folder, libs from build are inside lib, and not part of LDPATH unless installed and ldconfig is run.

thespianmark69, 2021-01-18:
I don't understand can you explain Sent from my Galaxy

DCKcode, 2021-01-19:
Thanks for the help guys - it was indeed inadvertently loading the system `libimobiledevice.so.6`.

```
LD_PRELOAD=$PWD/src/.libs/libimobiledevice-1.0.so.6:$LD_PRELOAD tools/idevicebackup2 backup ~/Desktop/
```
started the backup and did the trick. Sorry for the noise!

closed 2021-01-19 by DCKcode