libimobiledevice/libimobiledevice #1115: idevicebackup2 returns "Could not perform backup protocol version exchange,  error code -1" for IOS 14.3

jparris, 2021-01-27:
Hi there,

The iphone is paired with my Linux box and the non invasive commands work `idevicedate`, `ideviceinfo`,`idevicesyslog`, etc..

# Backup Output
```
idevicebackup2 --debug backup iphone_backup/
Backup directory is "iphone_backup/"
Started "com.apple.mobilebackup2" service on port 49732.
Could not perform backup protocol version exchange, error code -1
```

# Sample working command
```
idevicedate
Wed Jan 27 05:10:34 UTC 2021
```

# Version info
* Linux 
```
uname -a
Linux ... 5.4.73 #1-NixOS SMP Thu Oct 29 08:58:11 UTC 2020 x86_64 GNU/Linux
```
* IPhone SE running 14.3

mexmer, 2021-01-27:
you need to use latest version of idevicebackup and libimobiledevice, this has been already said multiple times.

#1112 , #1111, #1038, #1032, #1013, #1001

FYI github issue tracker has search function

jparris, 2021-01-27:
@mexmer Ah my confusion is that I was running the latest *released* version and instead I need to be running off of master.

closed 2021-01-27 by jparris

mexmer, 2021-01-28:
yes, many distribution packages are not updated yet, except experimental branches (like unstable debian)

mastertinner, 2021-02-27:
@mexmer, would it be possible to get a release soon that contains the fix for iOS 14?

mexmer, 2021-03-02:
it's up to your distribution maintainer, not up to me.

mastertinner, 2021-03-02:
How so? Any stable distribution will only update once there is a new GitHub release of this project which includes that code. So, in my view, creating a new versioned release of `libimobiledevice` is a prerequisite to then approach the distribution maintainers to add that new version to their package repositories.

nikias, 2021-03-02:
@mastertinner he is also not maintainer of this project. We will publish new releases soon.

mastertinner, 2021-03-02:
Sounds great. Sorry for mentioning you then, @mexmer and thanks for the update @nikias!

mastertinner, 2021-05-17:
@nikias, any news on the release? It would be great to see it soon as iOS has been released almost a year ago :)