libimobiledevice/libimobiledevice #1117: idevicedebugserverproxy and sys/select

sctol, 2021-02-02:
- [ ] The Feb 1, 2021 commit breaks building idevicedebugserverproxy under MSYS2/MINGW on Windows 10.  gcc cannot find sys/select.h, even though msys2-runtime-devel is installed.   I'm not sure what is going on.   The prior commit for libimobiledevice builds fine, and at least some of the tools work, such as idevicebackup2 and ideviceinfo.

- [ ] The dependency check for sys/select.h is also missing from configure*.

nikias, 2021-02-02:
Well, good thing is that it didn't even work before even though it might have compiled 🤣 -- at least with iOS >= 13.7.
I will check.

nikias, 2021-02-02:
Looks like it needs winsock2.h, will be basically what we have in iproxy:
https://github.com/libimobiledevice/libusbmuxd/blob/master/tools/iproxy.c#L40-L45

nikias, 2021-02-03:
Check again with commit 086e24e3808b7042532ea3d02c519da30f025728.

sctol, 2021-02-03:
It does build now.

A compiler warning is being thrown about windows.h appearing before winsock2.h

nikias, 2021-02-03:
Fixed with commit 5ef9f59.

closed 2021-02-03 by nikias

mexmer, 2021-02-04:
@nikias FYI winsock.h (or winsock2.h) is only header that always needs to be included before windows.h, other system headers should go after windows.h