libimobiledevice/libimobiledevice #1118: idevicedebug failed to get the task for process xxx

lnguyen234, 2021-02-09:
Is there anyone having this issue? I'm trying to start the app via idevicedebug on iOS13.5+ but always encounter that problem. The app starts but not fully be loaded and stuck at very first screen.
Thank you

supergithuber, 2021-05-17:
> Is there anyone having this issue? I'm trying to start the app via idevicedebug on iOS13.5+ but always encounter that problem. The app starts but not fully be loaded and stuck at very first screen.
> Thank you

same issue. Does any one find some solutions for this issue?

HMaker, 2023-06-10:
I have same issue, I guess it's caused by applications not built with debug mode enabled. Anyone fixed it?

nikias, 2023-06-10:
It will NEVER work with apps not built for debugging.