libimobiledevice/libimobiledevice #1120: Support killing a detached process

warpfactorfive, 2021-02-22:
The new idevicedebug detached option is very useful, but there is currently no mechanism to cleanup/kill the started process.  The process kill code already exists (see "kill process after we finish") so could it be added as an explicit command?

jpstotz, 2021-03-03:
But the command `debugserver_command_new("k", 0, NULL, &command);` only works because we heave previously connected to the debug-server and launched a process we are now connected to. 

Executing this code without launching the app before will not work. 

May be it will be easier for your use-case to use `frida-ps` and `frida-kill`.

warpfactorfive, 2021-03-03:
Thanks for the reply!  Perhaps the process ID can be returned by the call
to launch a detached process, and then passed back to the kill command?
I'm assuming that is enough information to re-establish the debugger
connection, but what do I know?  ;)