libimobiledevice/libimobiledevice #1121: Entering recovery mode on iOS 14.5 fails with code SessionInactive

a-gunter, 2021-02-26:
Was messing around with iOS 14.5 beta (18E5154f) and noticed that transitioning from normal mode -> recovery mode fails.

```
fm-700-58195:Beta ctdi$ ideviceenterrecovery -d -u [UDID]
16:12:29 lockdown.c:651 lockdownd_client_new(): device udid: [UDID]
Telling device with udid [UDID] to enter recovery mode.
16:12:29 lockdown.c:1061 lockdownd_enter_recovery(): telling device to enter recovery mode
16:12:29 property_list_service.c:132 internal_plist_send(): sending 297 bytes
16:12:29 service.c:144 service_send(): sending 4 bytes
16:12:29 idevice.c:600 idevice_connection_send(): internal_connection_send 4, sent 4
16:12:29 service.c:144 service_send(): sending 297 bytes
16:12:29 idevice.c:600 idevice_connection_send(): internal_connection_send 297, sent 297
16:12:29 property_list_service.c:137 internal_plist_send(): sent 297 bytes
16:12:29 property_list_service.c:138 internal_plist_send(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceenterrecovery</string>
	<key>Request</key>
	<string>EnterRecovery</string>
</dict>
</plist>
16:12:29 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
16:12:29 property_list_service.c:211 internal_plist_receive_timeout(): 292 bytes following
16:12:29 property_list_service.c:224 internal_plist_receive_timeout(): received 292 bytes
16:12:29 property_list_service.c:253 internal_plist_receive_timeout(): printing 292 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Error</key>
	<string>SessionInactive</string>
	<key>Request</key>
	<string>EnterRecovery</string>
</dict>
</plist>
16:12:29 lockdown.c:193 lockdown_check_result(): ERROR: SessionInactive
Failed to enter recovery mode.
```

Changing `lockdownd_client_new` to `lockdownd_client_new_with_handshake` fixes the issue.
https://github.com/libimobiledevice/libimobiledevice/blob/4837527745d98b9314eec1a250e2e13ce1ec3031/tools/ideviceenterrecovery.c#L98

&

https://github.com/libimobiledevice/idevicerestore/blob/7fd59d571648b0b418b4c3728b0fb64898feba8c/src/normal.c#L224

nikias, 2021-03-05:
Forgot to push this. See commit 25059d4c7d75e03aab516af2929d7c6e6d4c17de.

nikias, 2021-03-05:
For idevicerestore see https://github.com/libimobiledevice/idevicerestore/commit/6d3b6bba9127c02fc312468e4174e8148bf87472.

mexmer, 2021-03-05:
just quick question here @nikias , can session inactive happen anywhere else? i have not seen that error as long as i use libimobiledevice library for ios communication (or own implementation for windows).

nikias, 2021-03-05:
@mexmer haven't seen it anywhere else either

closed 2021-03-11 by a-gunter