libimobiledevice/libimobiledevice #1123: Would you wanna release 1.3.1 to public?

karthikk007, 2021-03-03:


karthikk007, 2021-03-03:
@nikias

jpstotz, 2021-03-03:
Based on your GitHub profile I would assume that you are an MacOS user.  On MacOS you can simply use the latest version from this GIT repo via brew:

In my opinion this is way better that waiting for a new release (which happens very very infrequent).

```
brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies ideviceinstaller
brew uninstall --ignore-dependencies usbmuxd
brew uninstall --ignore-dependencies libplist
sudo rm /var/db/lockdown/*
brew install --HEAD libplist
brew install --HEAD usbmuxd
brew unlink usbmuxd
brew link usbmuxd
brew install --HEAD libimobiledevice
brew install --HEAD ideviceinstaller

```

karthikk007, 2021-03-03:
And I am curious to know the anticipated date of release or release strategy if you know any.

nikias, 2021-03-03:
@jpstotz you should not do that
```
sudo rm /var/db/lockdown/*
```

jpstotz, 2021-03-04:
> @jpstotz you should not do that
> 
> ```
> sudo rm /var/db/lockdown/*
> ```

@nikias Thanks for your suggestion. I added it to prevent certain problems as described [here](https://github.com/libimobiledevice/libimobiledevice/issues/717#issuecomment-430622355). I already encountered such problems several times.

Could you please give some details why this command should not be used?

mexmer, 2021-03-05:
well, system might not like it, also by removing it, computer need to regen it's uid. if you have one iOS device, it might be good for you, if you have multiple iOS device, you will loos pairing with all of them, than includes wifi sync (if you enabled it).

you should use 
```
idevicepair unpair 
```
to remove pairing recoding for specific device that has problems and then pair it again.

nikias, 2021-03-05:
Thanks @mexmer for explaining this :)

SEbbaDK, 2021-03-19:
A tag for 1.3.1 for the iOS 14 fixes would be nice for releasing on package managers, but i understand if it will come in a later release.

stapelberg, 2021-08-15:
I was just trying to use `idevicebackup2` for the first time and encountered https://github.com/libimobiledevice/libimobiledevice/issues/1111. Then I noticed that while Arch Linux ships 1.3.0 (latest public release), the devs are telling people to update to an unreleased version (1.3.1) instead.

Could you please tag a new release so that distributions will ship the required changes for this program to work? I want to avoid hand-managing/compiling software from git :)

Thanks in advance!

PengJiang520, 2021-12-06:
> sudo rm /var/db/lockdown/*

sudo rm /var/db/lockdown/*
zsh: no matches found: /var/db/lockdown/*

PengJiang520, 2021-12-06:
> @jpstotz you should not do that
> 
> ```
> sudo rm /var/db/lockdown/*
> ```
no matches found: /var/db/lockdown/*

nikias, 2021-12-06:
PLEASE PLEASE do not try to remove anything in `/var/db/lockdown` EVER

baurmatt, 2022-05-14:
You can get the latest version (last commit from default branch) on Arch Linux through AUR libimobiledevice-git:

```bash
yay --aur libimobiledevice-git
```

Still, I'd like to support the request of a new release! It has been almost 2 years since the last one and iOS 14 is also quite old (and superseded by iOS 15) by now.

stephanbogner, 2024-08-20:
@nikias What would the correct way to handle it without "remove /var/db/lockdown"? Basically a corrected version of @jpstotz's command. Would probably help a lot of (less techy) folks, such as me. Thanks ❤️

mexmer, 2024-08-21:
> @nikias What would the correct way to handle it without "remove /var/db/lockdown"? Basically a corrected version of @jpstotz's command (for copy-pasting). Would probably help a lot of (less techy) folks, such as me. Thanks ❤️

if you call
_idevicepair unpair UDID_
usbmuxd will also remove lockdown pairing record for specified device UDID