libimobiledevice/libimobiledevice #1126: USB 2.0 speeds on iPad Pro 2020?

yet-another-user-eN3oh, 2021-03-08:
I'm transferring data to my iPad Pro (2020) via USB-C, and from lsusb I can see it's connected as a SuperSpeed+ (10 Gbps) device. However, the transfer rate is around 35 MB/s, which is more in line with the rates of USB 2.0. Is this a known limitation of the protocol? I'm also getting the same speeds under iTunes on Windows.

nikias, 2021-03-08:
The device doesn't handle more than USB 2.0 speeds.

badger200, 2021-04-17:
> The device doesn't handle more than USB 2.0 speeds.

Omg are you serious?? Why did they bother with USB-C at all then!?

nikias, 2021-04-17:
@badger200 quick charging!

badger200, 2021-04-17:
@nikias I can’t fathom restoring a 1TB iPad over USB 2.0.. It’s bad enough on my 256GB iPad. 

Utterly pathetic and shameful on Apple’s part. USB 3 has been standard equipment on all PCs and devices for what, 10 years now?

nikias, 2021-04-17:
Well to be fair that's my personal experience, but I never got more than ~30-35 MB/s transfer speeds even with USB-C between a MacBook Pro and an iPhone. For a 100 MB file transfer via AFC I get these results:
```
afc_file_write:
	SUCCESS: duration: 4.57 seconds size: 104857600 bytes, speed: 21.9 MB/s

afc_read_file:
	SUCCESS: duration: 3.09 seconds size: 104857600 bytes, speed: 32.4 MB/s
```

nikias, 2021-04-17:
Maybe on the USB-C iPad pro it is actually faster, I don't have such a device. That one actually has USB-C even on the device side, so maybe it's faster?

yet-another-user-eN3oh, 2021-04-17:
Well, the iPad Pros do support USB 3.1 (Gen2?). It shows up as an xhci device supporting SuperSpeed+ (10Gbps).

s1ris, 2021-04-18:
Good to see I'm not the only person with USB-C issues on their iPad. I have an iPad Air 4 which was marketed as having a 5gbps USB-C port (https://www.apple.com/newsroom/2020/09/apple-unveils-all-new-ipad-air-with-a14-bionic-apples-most-advanced-chip/). Apple rarely seems to say what transfer speeds their devices have over USB, but this is concrete. My device never goes beyond USB 2.0 spec speeds, and after reading bus info on both Mac and Windows, the iPad only ever gets a 2.0 connection.

A person I was discussing this issue with noticed Apple never specifically says that 5gbps is supported when connecting to a computer. "iPad Air now features a USB-C port for up to 5Gbps data transfer, which is 10 times faster, for connecting to cameras, hard drives, and external monitors up to 4K." Surely it's fair to say most people would assume that of course you can get that full 5gbps when connected via a Thunderbolt 3 cable (or some other C cable that supports the right speed) to your computer, right?

Sadly, based on what I've tested, it might be true. I was hoping to see if libimobiledevice could be used to get the actual full speed of what my iPad can do, but alas. Maybe this limitation is in the device firmware itself.

Funnily enough, in this (https://discussions.apple.com/thread/252549312) thread, a user is able to bypass what USB can do by using Ethernet adapters.

mexmer, 2021-04-19:
from my experience, even latest ipad pro doesn't support USB 3 superspeed for data transfer, we tried with external USB-C SSD and still got transfer roughly 50MB/s from/to ipad pro, while same SSD has troughput of 400MB/s with macbook pro. 

from that i can assume, while ipad pro has maybe gen 3.2 hardware connection, for commucation it still might use older protocols.