libimobiledevice/libimobiledevice #1128: [Feature Request] Attach to process

Spidy123222, 2021-03-11:
Hi I am spidy123222 and I am requesting this feature because I heard of a method that can attach a process when it launches the app and having to use remote debugserver seems less reliable and would like to have "Attach To Process".This would be debugging a app I would be using often called UTM and DolphinIOS where it uses a debugger to do JIT on the app. It has been found that on Xcode you can attach to process and make it able to work without being tethered via a cable.

jkcoxson, 2021-03-11:
tldr; we want to attach the debugger to a PID, like you can in Xcode 

Looking at idevicedebug it looks like this could be possible to contact the iOS debugger for a specific process, if you could point us in the right direction or to some documentation that would be great!

Spidy123222, 2021-03-12:
From going through it seems you can use udid to activate debugger and attach it to app bundle. But doing it over WiFi ain't supported it seems on Linux.

closed 2021-03-12 by Spidy123222