libimobiledevice/libimobiledevice #1129: WiFi sync on Ubuntu

jkcoxson, 2021-03-11:
Hello! I am wondering if WiFi sync is working on Ubuntu. There are many issues on this repo that have talked about WiFi sync, and I am wondering if this is possible or if it uses iTunes to WiFi sync, and therefore wouldn’t work on Ubuntu. If the protocol was completely reverse engineered, how would one go about using it? Thanks!

lostmaniac, 2021-03-30:
I am also looking for how to use

jkcoxson, 2021-03-30:
Tihmstar has a fork with Ubuntu WiFi sync. I haven't had any luck with it yet, but it's worth a shot.

jkcoxson, 2022-04-19:
I've created an extension/replacement for usbmuxd that will allow WiFi sync on Ubuntu called [netmuxd](https://github.com/jkcoxson/netmuxd). I can use AltServer and any other libimobiledevice tool over WiFi using it.

closed 2022-04-19 by jkcoxson