libimobiledevice/libimobiledevice #1131: `lockdownd_remove_value` not functioning

Nutr1t07, 2021-03-14:
Executing `lockdownd_remove_value(lockdown, "com.apple.purplebuddy", "AnimateLanugageChoice")` returns `LOCKDOWN_E_SUCCESS`, but the node is still remained.

I want to completely remove the node `AnimateLanugageChoice`. Is that really what I want to call `lockdownd_remove_value`? [The description of the function](https://github.com/libimobiledevice/libimobiledevice/blob/bfaf1a132a0d528cfe1dae8cbb4f0ecdeccaa88d/include/libimobiledevice/lockdown.h#L191) is `Removes a preference node by domain and/or key name.` I am not sure if the description is wrong.

<details>
  <summary>C Code</summary>

```
#include <libimobiledevice/libimobiledevice.h>
#include <libimobiledevice/lockdown.h>
#include <plist/plist.h>
#include <stdlib.h>
#include <stdio.h>

int main(){
  static char *udid = NULL;
  idevice_t device = NULL;

  if (idevice_new_with_options(&device, NULL, IDEVICE_LOOKUP_USBMUX) != IDEVICE_E_SUCCESS) {
    printf("ERROR: No device found!\n");
    return -1;
  }
  
  if (idevice_get_udid(device, &udid) != IDEVICE_E_SUCCESS) {
    printf("ERROR: Unable to get the device UDID.\n");
    idevice_free(device);
    return -1;
  }
  
  printf("Connected with UDID: %s\n", udid);
  
  lockdownd_client_t lockdown = NULL;
  lockdownd_client_new_with_handshake(device, &lockdown, "myapp");
  
  lockdownd_remove_value(lockdown, "com.apple.purplebuddy", "AnimateLanugageChoice");
  
  lockdownd_client_free(lockdown);
  idevice_free(device);
  free(udid);
  return 0;
}

```
</details>

Nutr1t07, 2021-03-14:
It seems that any nodes in `com.apple.purplebuddy` is not removable. It is still able to remove nodes from other domains.

mexmer, 2021-03-14:
that's not bug in libimobile but rather limitation of iOS. some values are immutable, but that is beyond scope of this library to know, which they are. library simply provides interface to call device function, if upon removal request device returns ok, operation did succeed. keep in mind, that some settings might have **default value**, that means, if you call remove it will remove custom value and revert to hardcoded default, that is most likely what you see, when trying to manipulate purplebuddy. again it's out of scope of library, because it's inside iOS.

Nutr1t07, 2021-03-14:
Perhaps it is. Anyway, you can still create new nodes and modify values of existing node in `com.apple.purplebuddy`.

nikias, 2021-03-14:
There's nothing wrong with the library, it's the device returning success while not removing the value. Nothing we can do here.

closed 2021-03-14 by nikias