libimobiledevice/libimobiledevice #1134: Continuous integration and publishing builds across platforms?

nisargjhaveri, 2021-03-28:
Can we setup CI/CD pipelines to verify and publish builds continuously?

The release cycle is slow for the project, and from the commit history, I can see that we see some Windows specific build breaks now and then. We should be able to help avoid that with some automatic build validations.

Also, generating builds from source is not usable for everyone. I know this is supposed to be a library, which would only be useful for people who are building stuff on top of this, but we also have some useful tools which can be used standalone by end-users.

Github actions may come handy. I see that we already have something setup for libplist. How can we extend that?

nikias, 2021-03-28:
I already started that with GitHub actions (on my personal repos, and will be deployed soon.

nisargjhaveri, 2021-03-28:
That's great! If needed, I would be happy to help out. :)