libimobiledevice/libimobiledevice #1135: idevicebackup2 hangs after mb2_receive_filename: too large filename length

kortschak, 2021-03-30:
I have just built libimobiledevice from source at the following commits.

libimobiledevice [master]$ git rev-parse HEAD
25059d4c7d75e03aab516af2929d7c6e6d4c17de
libusbmuxd [master]$ git rev-parse HEAD
3eb50a07bad4c2222e76df93b23a0161922150d1
libplist [master]$ git rev-parse HEAD
ebf2fdb938c755026668c8418da290e6870c1987

I am attempting to backup an iPad using idevicebackup2

```
$ idevicebackup2 --full backup .
Backup directory is "."
Started "com.apple.mobilebackup2" service on port 49997.
Negotiated Protocol Version 2.1
Starting backup...
Enforcing full backup from device.
Backup will be unencrypted.
Requesting backup from device...
Full backup mode.
<much snip>
```

after most of the work being done the program hangs after this.

```
[================================================  ]  95% Finished
[================================================  ]  95% Finished
[================================================  ]  95% Finished
Receiving files
Found new flag 06
ERROR: mb2_receive_filename: too large filename length (1315905651)!
[================================================  ]  95% Finished
```

(I agree with the exclamation mark in the error.)

ISTM that this is likely a corruption on the tablet, but the hang seems be due to libimobiledevice. Any hints about what might be the cause of the corruption would be greatly appreciated though not expected.

nikias, 2021-03-30:
Hmm `Found new flag 06` seems like what is causing some issue. I need to figure out what that flag is used for...

mexmer, 2021-04-01:
@nikias in my code, i skip code 6, not sure what it does either, but usually, when returning error from local computer to device, i set code 6 (that's what i found itunes does) but then again, it can have different meaning, when retrieving data from device, but so far, by ignoring this error, nothing did happen and backups are complete.

nikias, 2021-04-07:
@kortschak I just realized code 6 is already defined: `CODE_ERROR_LOCAL 0x06`
Seems like it wants to print an error message.
Can you try with this patch?
```diff
diff --git a/tools/idevicebackup2.c b/tools/idevicebackup2.c
index 068420d..142824f 100644
--- a/tools/idevicebackup2.c
+++ b/tools/idevicebackup2.c
@@ -1080,7 +1080,7 @@ static int mb2_handle_receive_files(mobilebackup2_client_t mobilebackup2, plist_
                }

                /* TODO remove this */
-               if ((code != CODE_SUCCESS) && (code != CODE_FILE_DATA) && (code != CODE_ERROR_REMOTE)) {
+               if ((code != CODE_SUCCESS) && (code != CODE_FILE_DATA) && (code != CODE_ERROR_REMOTE) && (code != CODE_ERROR_LOCAL)) {
                        PRINT_VERBOSE(1, "Found new flag %02x\n", code);
                }

@@ -1135,7 +1135,7 @@ static int mb2_handle_receive_files(mobilebackup2_client_t mobilebackup2, plist_
                }

                /* check if an error message was received */
-               if (code == CODE_ERROR_REMOTE) {
+               if (code == CODE_ERROR_REMOTE || code == CODE_ERROR_LOCAL) {
                        /* error message */
                        char *msg = (char*)malloc(nlen);
                        mobilebackup2_receive_raw(mobilebackup2, msg, nlen-1, &r);
```

kortschak, 2021-04-07:
Thanks, I'll try that when I have the device here again and report back.

kortschak, 2021-04-08:
The remote error appears to have been "Received an error message from device: No such file or directory (2)". This was reported a number (94) of times. It could potentially be helpful to report the file name in these cases, or is that not known/not useful?

It does then move on to moving files, but appears to hang after this
```
<snip>
Moving 128 files
[==================================================] 100% Finished
Moving 128 files
[==================================================] 100% Finished
Moving 57 files
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
```
Sitting on 100% CPU for over half an hour before I cancelled it.

mexmer, 2021-04-09:
@kortschak apple backup is controlled by device, why your computer sits at 100% i have no idea, unless there is some loop in code (or your cpu is weak), it should not be related. as for that error message it means exactly what it means. while internally backup service copy some files to temporary folder before sending them to computer, it also copies files directly from application data folder, it's possible that application did some cleanup since start of backup.

itunes backup first does "snapshot" of what needs to be backed up, then sends files to computer, although missing 94 files seems to me quite much.

also what puzzles me, that in snip i see move file operation, is this prebackup phase? if so, did you remove any files from your itunes backup folder?

kortschak, 2021-04-09:
Thanks, @mexmer. The linux box is not weak and the iPad is new, so those are likely not a cause for the problem. The moving under foot makes sense (I see that in my rsync backup of my work stations.

I'm not sure what the phases of the back up are; it successfully completes the receiving of files and then does a lot of file moving as shown. This is the first attempt at backing up the iPad. I don't have an iTunes folder (no mac) and the device itself is not mine (family member) so I don't touch any files on it.

mexmer, 2021-04-12:
apple backup (for quite some time already) first creates temporary folder in target location into which it stores files received from device, after all files are received, including manifest file, backup moves files from temporary folder into main folder (which you have specified as location for backup). when all files are moved, temporary folder is erased (since it move operation, it should be empty) and into status.plist "Completed" is written, also info.plist is created.

nikias, 2021-04-12:
@kortschak @mexmer this is the final stage of the backup where it moves stuff from a `Snapshot` inside the backup directory to the final location. Are you using this on a normal filesystem? Or is this some mounted network share or something? You can also run `idevicesyslog -p backupd -p BackupAgent2` in another terminal to get some backup-only related debug output, maybe that reveals something.

kortschak, 2021-04-13:
It's a local filesystem on the machine running idevicebackup2.

I have run `idevicesyslog -p backupd -p BackupAgent2` and it finishes up with:

```
<snip>
Apr 13 12:00:44 BackupAgent2[3895] <Notice>: Finished uploading 2 batches, s:{0,0,0,0}, t:1.062, r:0.000
Apr 13 12:00:46 BackupAgent2[3895] <Notice>: ManifestDB: closing /var/root/Library/Caches/Backup/58EDE7C7-461F-4AD7-B71E-E40E74F6E0D4/Manifest.db
Apr 13 12:00:46 BackupAgent2[3895] <Notice>: ManifestDB: opening /var/root/Library/Caches/Backup/58EDE7C7-461F-4AD7-B71E-E40E74F6E0D4/Manifest.db
Apr 13 12:00:50 BackupAgent2[3895] <Notice>: Uploaded 230 files of 30.7 MB in 6.934 s at 4.44 MB/s
Apr 13 12:01:19 BackupAgent2(Foundation)[3895] <Notice>: *** Assertion failure in -[MBDriveScript performingOperation:], MBDriveScript.m:71
Apr 13 12:01:19 BackupAgent2(CoreFoundation)[3895] <Notice>: *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Deviating from script (expecting MoveItems, got UploadFiles)'
*** First throw call stack:
(0x189dbf9d8 0x19e142b54 0x189cce50c 0x18afc8878 0x100362b70 0x100363650 0x10031e0ec 0x10031e324 0x10031e798 0x1003332a0 0x10032f320 0x100320814 0x100320080 0x1003203dc 0x100335bbc 0x100335714 0x1c5159f30 0x1c515aff8 0x189d1f530 0x189d40420 0x189d3f834 0x189d399f4 0x189d38ba0 0x189d39c60 0x100334a98 0x100334224 0x189a17568)
```

And at this point the backup hangs.

I have the full logs from the `idevicesyslog`, but won't paste them here. If there are particular things you need to know, please let me know.

mexmer, 2021-04-14:
@kortschak can you also change line 1144 in idevicebackup2.c from
```
			if (last_code != CODE_FILE_DATA) {
```
to
```
			if (last_code != CODE_FILE_DATA || code == CODE_ERROR_LOCAL) {
```
it's additional change to one that handles flag 6
because it seem it doesn't print back message that it had received from device on local error. maybe that will give us any insight to what is happening

kortschak, 2021-04-14:
With that change I see the same thing, the output as above (repeated here):
```
<snip>
Apr 14 20:28:57 BackupAgent2[5119] <Notice>: Finished uploading 2 batches, s:{0,0,0,0}, t:1.015, r:0.000
Apr 14 20:28:59 BackupAgent2[5119] <Notice>: ManifestDB: closing /var/root/Library/Caches/Backup/2CCA0091-8A22-452D-9682-01E3E9F87E6B/Manifest.db
Apr 14 20:28:59 BackupAgent2[5119] <Notice>: ManifestDB: opening /var/root/Library/Caches/Backup/2CCA0091-8A22-452D-9682-01E3E9F87E6B/Manifest.db
Apr 14 20:29:04 BackupAgent2[5119] <Notice>: Uploaded 230 files of 30.7 MB in 8.252 s at 3.73 MB/s
Apr 14 20:29:34 BackupAgent2(Foundation)[5119] <Notice>: *** Assertion failure in -[MBDriveScript performingOperation:], MBDriveScript.m:71
Apr 14 20:29:34 BackupAgent2(CoreFoundation)[5119] <Notice>: *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Deviating from script (expecting MoveItems, got UploadFiles)'
*** First throw call stack:
(0x189dbf9d8 0x19e142b54 0x189cce50c 0x18afc8878 0x102ccab70 0x102ccb650 0x102c860ec 0x102c86324 0x102c86798 0x102c9b2a0 0x102c97320 0x102c88814 0x102c88080 0x102c883dc 0x102c9dbbc 0x102c9d714 0x1c5159f30 0x1c515aff8 0x189d1f530 0x189d40420 0x189d3f834 0x189d399f4 0x189d38ba0 0x189d39c60 0x102c9ca98 0x102c9c224 0x189a17568)
```
and the backup hanging with these final lines
```
<snip>
[==================================================] 100% Finished
Moving 128 files
[==================================================] 100% Finished
Moving 90 files
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
Moving 1 file
[==================================================] 100% Finished
```

nikias, 2021-04-14:
So BackupAgent2 is crashing (terminating because of an exception), and idevicebackup2 keeps waiting for more data which never arrives.
The exception message makes it sound like it received UploadFiles instead of MoveItems:
`Deviating from script (expecting MoveItems, got UploadFiles)`
Pretty weird but maybe due to that "local error" things get messed up... I don't really understand yet what is happening...

mexmer, 2021-04-14:
@kortschak can you check syslog output if you revert that change with ERROR_CODE_LOCAL (both changes, mine and nikias).
you will probly see again "new flag 0x6", but we might see different error in syslog.

although i assume too, backup service crashes in phone.

on another note, from my observation, sometimes if backup from apple device repeatedly fails, restarting device helps.

kortschak, 2021-04-14:
Thank you both for all your help here. @nikias, yes, that's pretty much what I think is going on. The device is failing backups to osx hosts as well, and to cloud, so there is something very odd going on. Using idevicebackup2 has helped me understand at least where the issues are happening. @mexmer I will try that when the device is available again and report back, and yes, as mentioned above, the backups fail on Apple-native systems.

kortschak, 2021-04-19:
After many errors the syslog finishes up with this
```
<snip>
Apr 19 11:15:43 backupd[11870] <Error>: Failed to upload AppDomain-com.vblast.flipaclip:Library/projects/p199/l15/259483.fci (<CKError 0x1068e0a70: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">)
Apr 19 11:15:43 backupd[11870] <Error>: Operation B737F128AC358A0F/<MBCKBatchSave: 0x1060bd740> failed: <CKError 0x10628b700: "Partial Failure" (2/1011); "Failed to modify some records"; partial errors: {
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:yWogPjEKVjgvWHzXPhYkH+zgn4k=:13153:BSFF:(_defaultZone:__defaultOwner__) = <CKError 0x1061a0dd0: "Internal Error" (1/1000); "pread error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:I7KMJrPwptR1HIDn00pkpMa1284=:6718:Ki2F:(_defaultZone:__defaultOwner__) = <CKError 0x1061a0280: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:M/4t9NhQNadwrLwU/0iGYyYmHx0=:740:S39I:(_defaultZone:__defaultOwner__) = <CKError 0x1068d0df0: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:nEyG6cXUYGvTgpqSFeFiiLBPGZ0=:4075:Cl+Y:(_defaultZone:__defaultOwner__) = <CKError 0x1061aaed0: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">
	F:6C2
Apr 19 11:15:43 backupd[11870] <Error>: CKModifyRecordsOperation failed, id:B737F128AC358A0F, gid:7DB00D848CBCB80D, t:0.579s, o:5: <CKError 0x106279e20: "Partial Failure" (1011); "Failed to modify some records"; partial errors: {
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:yWogPjEKVjgvWHzXPhYkH+zgn4k=:13153:BSFF:(_defaultZone:__defaultOwner__) = <CKError 0x1061a0dd0: "Internal Error" (1/1000); "pread error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:I7KMJrPwptR1HIDn00pkpMa1284=:6718:Ki2F:(_defaultZone:__defaultOwner__) = <CKError 0x1061a0280: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:M/4t9NhQNadwrLwU/0iGYyYmHx0=:740:S39I:(_defaultZone:__defaultOwner__) = <CKError 0x1068d0df0: "Asset File Not Found" (16/3002); "open error: 2 (No such file or directory)">
	F:6C213366-44B7-401D-BBDB-5865E2C2A62C:nEyG6cXUYGvTgpqSFeFiiLBPGZ0=:4075:Cl+Y:(_defaultZone:__defaultOwner__) = <CKError 0x1061aaed0: "Asset File Not Found" (16/3002); "open error: 2
Apr 19 11:37:06 backupd[11870] <Notice>: Operation C7DB178B76DB897C/<MBCKBatchSave: 0x1060bd740> completed
Apr 19 11:37:06 backupd[11870] <Notice>: CKModifyRecordsOperation completed, id:C7DB178B76DB897C, gid:7DB00D848CBCB80D, t:1290.829s, o:4
^C
Exiting...
```

The re

mexmer, 2021-04-19:
it really looks like some files have been moved/deleted during backup. which normally should not happen, but if sync operation if app on phone happen, it can. having phone in flight mode, when doing backup is usually good idea.

nikias, 2021-04-19:
It is an app-specific file too, of the app `com.vblast.flipaclip`. Usually the backup system is able to restart if it detects changes, and I know for sure it works for system settings files but not sure if it also monitors application files. The problem might also be a symlink pointing to a file that doesn't exists. If flight mode doesn't help, remove that app and install it later if required.

kortschak, 2021-04-19:
@nikias The issue of a corrupted fs from an app was my thinking.

I've try the flight mode and while it still fails, the application-specific error lines are not present in the syslog.

Unfortunately $family_member will not let me try the second experiment.

kortschak, 2021-05-31:
The device has now been backed up and restored to cloud storage and subsequently the backup by `idevicebackup2` now succeeds. The underlying cause seems likely to have been file system corruption on the device, but we will never know.

Thank you for you time working through this.

closed 2021-05-31 by kortschak