libimobiledevice/libimobiledevice #1140: gnutls undefined reference to symbol

simonroy-h, 2021-04-10:
I disabled openssl.

Currently getting:

/usr/bin/ld: ../common/.libs/libinternalcommon.a(userpref.o): undefined reference to symbol 'gnutls_x509_crt_sign2@@GNUTLS_3_4'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libgnutls.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:771: idevicepair] Error 1

Not sure what to do here?

volinthius, 2021-04-12:
Looks like commit 735ac51e0eeaa3419e85dbad9579401ae828f631 is causing this. Going back to commit 25059d4c7d75e03aab516af2929d7c6e6d4c17de fixed build for me.

mexmer, 2021-04-12:
looks like version compatibility issue on you system

do you have installed same version of gnutls-dev and gnutls-bin package?
or do you have installed multiple version of gnutls-dev?

nkichukov, 2021-04-12:
I experienced the same, gnutls version is 3.6.15 here.
GNU/Gentoo Linux here on amd64.

mexmer, 2021-04-12:
well guessing from that messages, it wants GNU TLS version 3.4, that means it was compiled against 3.4 headers, while you have 3.6 installed.

from that you can guess, you have 3.4 development package installed, but your system default is 3.6, check your package manager remove all other versions of gnutls except 3.6 and do **./configure** again, then make clean, make all.

nikias, 2021-04-13:
Should be fixed with commit 4b2f17e8cce8fa078aa55f5da5f726c20eac3e2e.

nkichukov, 2021-04-13:
Confirmed, nikias patch fixes the problem for me. Thank you.

mexmer, 2021-04-14:
@nikias from you patch i assume gnutls split functionality into multiple libraries? i compile most stuff against openssl, so never bothered with checking gnutls structure.