libimobiledevice/libimobiledevice #1141: idevicesyslog, doesn't show debug/info messages in output

0x3c3e, 2021-04-12:
`os_log("MSG", type: .debug)` or `os_log("MSG", type: .info)` messages aren't visible in output even with `-d` flag.

Are there options like in Console.app to enable debug/info levels?

francybiga, 2021-07-16:
It would be very useful to access debug and info logs!

jay0129, 2021-12-09:
I need this feature so badly!

localip, 2022-06-12:
Bump, is this planned?

localip, 2022-06-12:
@libimobiledevice

ecunha-ta, 2023-03-14:
Any news on this?

sejinik, 2023-07-21:
is this planned?!  it will be helpful to user

lucasfernog, 2024-02-25:
I wish this had a solution too :(

Dev1an, 2025-06-26:
Yes I tried idevicesyslog but that only shows messages of type .default, .error or .fault
Is idevicesyslog part of libimobildevice or is there a separate repo for that?

Dev1an, 2025-06-26:
@nikias What is the "new updated idevicesyslog"? I tried `idevicesyslog 1.3.0`

nikias, 2025-06-26:
You need to compile it from source.

closed 2025-07-08 by 0x3c3e