libimobiledevice/libimobiledevice #1142: Possible crash during afc_read_directory

skortela, 2021-04-15:
I have seen sometimes crashes (windows env) when directory listing is ongoing and device is disconnected.

According to code, it could be that afc_read_directory returns AFC_E_SUCCESS and directory_information returns NULL. In this case caller app crashes as it assumes that directory list is received.

This could happen if afc_receive_data returns ok, with actual read bytes to be 0.
In that case make_strings_list returns NULL.

I'm not sure how afc_receive_data behaves, can it return AFC_E_SUCCESS with 0 bytes in some scenarios?

> 	/* Receive the data */
	ret = afc_receive_data(client, &data, &bytes); // ret =AFC_E_SUCCESS, bytes = 0
	if (ret != AFC_E_SUCCESS) {
		if (data)
			free(data);
		afc_unlock(client);
		return ret;
	}
	/* Parse the data */
	list_loc = make_strings_list(data, bytes); // list_loc = NULL
	if (data)
		free(data);

>	afc_unlock(client);
	*directory_information = list_loc;

>	return ret;
>}

mexmer, 2021-04-15:
yes, during data receive operation, you can receive empty packet.
actually i had phone, that did send empty directory listing (null data, not even . and ..), according to syslog, it was internally crashing on something ... restart resolved that problem.

as for receiving empty data, look here
https://github.com/libimobiledevice/libimobiledevice/blob/4b2f17e8cce8fa078aa55f5da5f726c20eac3e2e/src/afc.c#L248

but receiving empty data to dir list command is quite rare situation, it causes no problem with afc_read_dir function, if you look inside make_strings_list, function check for empty data.
https://github.com/libimobiledevice/libimobiledevice/blob/4b2f17e8cce8fa078aa55f5da5f726c20eac3e2e/src/afc.c#L384

so any function calling afc_read_dir should check for null value returned in directory_information parameter.

skortela, 2021-04-15:
I disagree that caller of afc_read_dir should check NULL value of directory_information, there is return code for those scenarios. And according documentation NULL is returned only on errorcases, so not if function returns AFC_E_SUCCESS.

```
/**
 * Gets a directory listing of the directory requested.
 *
 * @param client The client to get a directory listing from.
 * @param path The directory for listing. (must be a fully-qualified path)
 * @param directory_information A char list of files in the directory
 *        terminated by an empty string or NULL if there was an error. Free with
 *        afc_dictionary_free().
 *
 * @return AFC_E_SUCCESS on success or an AFC_E_* error value.
 */
afc_error_t afc_read_directory(afc_client_t client, const char *path, char ***directory_information);
```

So i would fix it in that way:
```
/* Receive the data */
ret = afc_receive_data(client, &data, &bytes);
if (ret != AFC_E_SUCCESS) {
    if (data)
        free(data);
    afc_unlock(client);
    return ret;
}
else if (bytes == 0) {
    // empty resp?
    debug_info("Received empty packet!");
    if (data)
        free(data);
    afc_unlock(client);
    return AFC_E_IO_ERROR;
}
```

mexmer, 2021-04-15:
you might disagree, but thing is device didn't send back error, it did send empty data, it can also happen during reading file for example,  (both id doing AFC_DATA_OPERATION), if it's resolved to error, it will interrupt file reception. 

i have also one more situation, when it returns no data, if you try to call read_directory on symlink, that you don't have access to target folder (or it doesn't exits), over afc it will not throw any error, it will just return empty directory listing (without . and ..).

so documentation needs to be modified rather than code. mind device returns to you exact ammount of data, that it says in packet length, so there is no error in communication. while you don't know why you had received empty data, treating empty data as error is wrong.

if there is error anywhere, it's inside apple own server implementation, because it sends empty data, but that again might be related to their protocol behavior, which we can only guess.

nikias, 2021-04-15:
I think the general problem is that the code doesn't detect the connection abort (which is what you see with the "empty data"), and in that case it also needs to return an error code. I need to see how I can improve this.
A similar issue is with the underlying code in idevicerestore which causes the `ERROR: Unable to receive message from FDR 0x7f05d8001320 (-2). 0/2 bytes` kind of error messages, which is actually just a "remote side closed the connection" event.
@skortela I also agree that the returned directory information should never be NULL when it returns success.

mexmer, 2021-04-15:
@nikias thing is, i have seen multiple times, that phone did send empty data, and didn't abort connection.

i agree tho' that any connection error should be propagated to upper levels.