libimobiledevice/libimobiledevice #1143: pbpaste and pbcopy commands

eadmaster, 2021-04-16:
Not sure if it is doable, but i think it would be very useful having a command to copy/paste strings from/to the device clipboard.

mexmer, 2021-04-16:
afaik there is no service in ios device for that..

eadmaster, 2021-04-17:
this seems to be possible via the [Universal Clipboard](https://support.apple.com/en-gb/guide/mac-help/mchl70368996/mac), but i'm still looking for a way to access than on linux...

mexmer, 2021-04-19:
universal clipboard is synchronized over internet, it is not done trough cable connection. it uses apple continuity service, what you put into UC is uploaded into apple cloud, then downloaded to other devices that use same account as your mac.

nikias, 2021-04-19:
@mexmer I don't think that is entirely true, continuity/handoff uses wifi direct connections and bluetooth, similar to AirDrop. This is a completely different area that libimobiledevice is not covering at this time.

captn3m0, 2024-03-05:
I'm not sure if it works without BT discovery entirely, but Airdrop now uses the Wired Connection for the newer iPhones (with USB-C). There's some user reports: [[1]](https://mastodon.social/@_inside/112043390973974454), [[2]](https://www.reddit.com/r/iphone/comments/1807654/airdrop_using_wired_connection/), but unclear if it will work with WiFi/BT disabled.