libimobiledevice/libimobiledevice #1144: Cannot write to file provider storage on elementary OS

hanaral, 2021-04-16:
This has been an issue for me since 2019(?) but I never bothered reporting it because I always thought Apple had permenently changed something that made it not work using libimobiledevice. However, now that I look at the gh it seems to just be me.
I've tested on both my iPhone(s) and iPad on iOS 13 and up, it always produces the same result: Copying and viewing file provider storage is fine, but trying to write results in nothing. 
Mounting the file provider storage from command line results in the file manager crashing or redirecting to another drive when trying to view the files. I have not been able to ascrertain what specifically goes wrong with the writing but I would guess it has something to do with afc2

platform: elementary OS 6 Early Access (Ubuntu 20.04)
libimobiledevice version: latest availabe(libimobiledevice-dev/focal 1.2.1~git20191129.9f79242-1build1 i386)

mexmer, 2021-04-19:
afc2 is available only on jailbroken phones, but it doesn't give you access to delete or write everywhere. technically you are having root access trough afc2, but still protected, opened or locked files, also encrypted files (even phone is locked) might not be accessible either for read or write. some locations are also readonly inside phone, unless you remount them with readwrite access.

i tested libimobile + ifuse with afc2 and for wiriting into media folder, or logs it works as it should, i didn't want to mess with my phone, so didn't try to write to other locations, but it should be limited only by access privileges.

mind protocol wise there is no difference between AFC and AFC2 service, difference is only in access, AFC is limited only to media folder sandbox, while AFC2 is without sandbox (full device filesystem)

hanaral, 2021-04-19:
I'm aware of the differences in write privileges between jailbroken and non-j.b. devices. My issue is that the file provider storage from different apps(such as Documents by Readdle) are not writable, despite them supposedly being able to be written to by design. Even the apps' folder icons are correctly changed to their individual icons, as it was in the past when it worked.
There is no way this is intentionally broken, so if AFC isn't the culprit then maybe the way files are transferred to iOS 13+ devices

mexmer, 2021-04-19:
if you mount filesharing from application, for most application you need to have iOS device unlocked to write into filesharing folder.

afaik there was no change to filesharing since they implemented restictions in 8.3, and that restriction was that you can write to/read from only documents folder and Library and tmp are inaccessible.

just tested it in our app, that is based on libimobile on iphone7 with ios 14.4

hanaral, 2021-04-19:
Yes, all the devices have tested with have been unlocked and paired when transferring - it hasn't made any difference. Which app have you written to from your computer?

mexmer, 2021-04-20:
VLC, Acrobat (Reader), Pages, (Microsoft) Word.

can you check on computer with windows/macos if you can write using document sharing features, on windows this is done trough itunes on macos depends on your os version in 11 and 10.15 (catalina) it's native in finder, in older also itunes like windows.