libimobiledevice/libimobiledevice #1147: idevicebackup2 unback: option to skip files to workaround crashes in BackupAgent2

pabs3, 2021-04-27:
With an old iPad2 I have, `BackupAgent2` crashes on the device when attempting to unback some files, with the error `Domain has no root path (uninstalled app?): SysSharedContainerDomain-systemgroup.com.apple.osanalytics`.

I think if idevicebackup2 unback could skip this file, then this would workaround the crash by moving to the next file and continuing from there.

Alternatively, when encountering what looks like a `BackupAgent2` crash, unback could reset the connection and try to start from the next file, this would help users who don't know about idevicesyslog.

I've included some debugging information below so you can confirm my assessment of this issue.

```
$ idevicebackup2 unback idevicebackup2-backup/
Backup directory is "idevicebackup2-backup/"
Started "com.apple.mobilebackup2" service on port 62560.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting to unpack backup...
Backup password: No
Sending 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Manifest.plist' (8.5 KB)
Sending 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Manifest.mbdb' (335.5 KB)
Copying 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' to '_unback_/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/var/mobile/Applications/com.apple.Maps/Library/Preferences/com.apple.Maps.plist'
Copying 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' to '_unback_/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/var/mobile/Applications/com.apple.Maps/Library/Maps/Pins.mapsdata'
Copying 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' to '_unback_/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/var/mobile/Applications/com.apple.Maps/Library/Maps/History.mapsdata'
Copying 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' to '_unback_/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/var/mobile/Applications/com.apple.Maps/Library/Maps/GeoHistory.mapsdata'
Copying 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' to '_unback_/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/var/mobile/Applications/com.apple.Maps/Library/Maps/FailedSearches.mapsdata'
ERROR: Could not receive from mobilebackup2 (-256)
Unback Aborted.
$ echo $?
255
```

```
$ idevicesyslog
...
Apr 27 22:11:12 kernel[0] <Notice>: AppleKeyStore:Sending lock change 0 for handle 0
Apr 27 22:11:12 UserEventAgent[23] <Error>:  LockStateNotifier aksNotificationCallback posting notification: com.apple.mobile.keybagd.lock_status
Apr 27 22:11:12 UserEventAgent[23] <Notice>: (Note ) PIH: Lock status changed.
Apr 27 22:11:12 kernel[0] <Notice>: AppleKeyStore:Sending lock change 0 for handle 0
Apr 27 22:11:12 UserEventAgent[23] <Error>:  LockStateNotifier aksNotificationCallback posting notification: com.apple.mobile.keybagd.lock_status
Apr 27 22:11:12 UserEventAgent[23] <Notice>: (Note ) PIH: Lock status changed.
Apr 27 22:11:12 BackupAgent2[6740] <Warning>: INFO: Loading system domains
Apr 27 22:11:12 BackupAgent2[6740] <Warning>: INFO: Loaded backup rules for 'KeyboardDomain' from external plist at '/System/Library/Backup/KeyboardDomain.plist'
Apr 27 22:11:12 BackupAgent2[6740] <Warning>: INFO: Backup properties: <MBDriveProperties: version=9.1, systemDomainsVersion=24.0, date=2021-04-27 13:30:25 +0000, encrypted=0, passcodeSet=0, lockdownKeys={UniqueDeviceID: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX", com.apple.TerminalFlashr: {}, com.apple.mobile.data_sync: {Notes: {AccountNames: ["iCloud"], Sources: ["iCloud"]}, Calendars: {AccountNames: ["iCloud"], Sources: ["iCloud"]}, Contacts: {AccountNames: ["iCloud"], Sources: ["iCloud"]}, Bookmarks: {AccountNames: ["iCloud"], Sources: ["iCloud"]}}, BuildVersion: "13G36", com.apple.mobile.iTunes.accessories: {}, DeviceName: "XXXXXXXXXXXXX", com.apple.mobile.wireless_lockdown: {}, com.apple.Accessibility: {SpeakAutoCorrectionsEnabledByiTunes: 0, MonoAudioEnabledByiTunes: 0, VoiceOverTouchEnabledByiTunes: 0, InvertDisplayEnabledByiTunes: 0, ZoomTouchEnabledByiTunes: 0, ClosedCaptioningEnabledByiTunes: 0}, com.apple.MobileDeviceCrashCopy: {}, SerialNumber: "XXXXXXXXXXXX", ProductType: "iPad2,2", ProductVersion: "9.3.5"}>
Apr 27 22:11:12 BackupAgent2[6740] <Warning>: INFO: Backup database: <MBDatabase: version=5.0>
Apr 27 22:11:12 BackupAgent2[6740] <Warning>: *** Assertion failure in -[MBDomain rootPath], /BuildRoot/Library/Caches/com.apple.xbs/Sources/DeviceBackup/MobileBackup-1472.15/Versions/2/Source/MBDomain.m:539
Apr 27 22:11:12 BackupAgent2[6740] <Error>: *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Domain has no root path (uninstalled app?): SysSharedContainerDomain-systemgroup.com.apple.osanalytics'
	*** First throw call stack:
	(0x2426791b 0x23a02e17 0x242677f1 0x24a4b529 0x141f8f 0x123439 0x12c74b 0x10ea05 0x2d8db88b 0x2d8dc541 0x24217d4f 0x2422a5b3 0x24229cc7 0x242280d9 0x24177229 0x241b7dd3 0x10d3e5 0x10ce79 0x23e1f873)
Apr 27 22:11:12 com.apple.xpc.launchd[1] (lockdown.96906059338368.com.apple.mobilebackup2[6740]) <Notice>: Service exited due to signal: Abort trap: 6
Apr 27 22:11:12 ReportCrash[6741] <Error>: assertion failed: 13G36: libsystem_trace.dylib + 15927 [E82A6F2D-873A-39AD-8014-EDEB52248157]: 0x0
Apr 27 22:11:12 Unknown[6741] <Error>: 
Apr 27 22:11:12 ReportCrash[6741] <Warning>: os_activity_diagnostic_for_pid() failed!
Apr 27 22:11:12 ReportCrash[6741] <Notice>: Formulating report for corpse[6740] BackupAgent2
Apr 27 22:11:12 ReportCrash[6741] <Warning>: Saved type '109(109_BackupAgent2)' report (22 of max 25) at /var/mobile/Library/Logs/CrashReporter/BackupAgent2-2021-04-27-221112.ips
```

```
$ idevicebackup2 list idevicebackup2-backup/ | grep osanalytics
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX,                                         , 040755,    1541,   0,   0, 2019-11-14 10:33:09, 2019-11-14 10:33:27, 2019-11-14 10:33:09,          0,     0, SysSharedContainerDomain-systemgroup.com.apple.osanalytics, , , {}
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX,                                         , 040000,    1544,  -2,  -2, 2019-11-14 10:33:09, 2019-11-14 10:33:27, 2019-11-14 10:33:09,          0,     0, SysSharedContainerDomain-systemgroup.com.apple.osanalytics, Library, , {}
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX, XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX, 100600, 3149046, 501, 501, 2021-04-27 08:27:30, 2021-04-27 08:27:30, 2021-04-27 08:27:27,       4663,     4, HomeDomain, Library/Preferences/com.apple.osanalytics.addaily.plist, , {}
```

```
$ idevicecrashreport -k idevicecrashreport/ > /dev/null
$ cat idevicecrashreport/BackupAgent2-2021-04-27-221112.ips
{"app_name":"BackupAgent2","share_with_app_devs":false,"name":"BackupAgent2","app_version":"","is_first_party":true,"os_version":"iPhone OS 9.3.5 (13G36)","slice_uuid":"xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx","bug_type":"109","build_version":"","timestamp":"2021-04-27 22:11:12.12 +0800"}
Incident Identifier: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
CrashReporter Key:   xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Hardware Model:      iPad2,2
Process:             BackupAgent2 [6740]
Path:                /usr/libexec/BackupAgent2
[... 189 lines cut ...]
0x2a901000 - 0x2a921fff GSS armv7  <2059afd7dc273df2a6162a6cf671c576> /System/Library/Frameworks/GSS.framework/GSS
0x2a971000 - 0x2a979fff FMCoreLite armv7  <767fa8f1605f3aa6b836d280b601608f> /System/Library/PrivateFrameworks/FMCoreLite.framework/FMCoreLite
0x2a97a000 - 0x2a97dfff MessageSupport armv7  <76826157d32531158f2d44dd37b655de> /System/Library/PrivateFrameworks/MessageSupport.framework/MessageSupport
0x2a97e000 - 0x2a9befff MIME armv7  <df86b75f60ca30b794bdf5cb63ed2736> /System/Library/PrivateFrameworks/MIME.framework/MIME
0x2a9bf000 - 0x2a9c4fff AOSNotification armv7  <f917d58b5754386098dbb21fbb9c4388> /System/Library/PrivateFrameworks/AOSNotification.framework/AOSNotification
0x2a9c5000 - 0x2aa1ffff AccountsDaemon armv7  <d183094acb9f337587a14299ae51625c> /System/Library/PrivateFrameworks/AccountsDaemon.framework/AccountsDaemon
0x2aa20000 - 0x2aa3dfff AppleIDSSOAuthentication armv7  <f353046f873931af92cb64cea789af81> /System/Library/PrivateFrameworks/AppleIDSSOAuthentication.framework/AppleIDSSOAuthentication
0x2aa3e000 - 0x2aa4cfff MailServices armv7  <a20960d220243ad39188eb8b918bdc48> /System/Library/PrivateFrameworks/MailServices.framework/MailServices
[... cut ...]
0x2df48000 - 0x2df53fff SetupAssistant armv7  <bc10a3ac806c3433b5bd73128fea7951> /System/Library/PrivateFrameworks/SetupAssistant.framework/SetupAssistant
0x2e14f000 - 0x2e15bfff FindMyDevice armv7  <0effb29cb33539b494be5a83bf06bdd6> /System/Library/PrivateFrameworks/FindMyDevice.framework/FindMyDevice
0x37f8d000 - 0x37f95fff libMobileGestaltExtensions.dylib armv7  <f7c090a63ee73d8cad7c7a860efbc75f> /usr/lib/libMobileGestaltExtensions.dylib

Error Formulating Crash Report:
Failed while requesting activity/breadcrumb diagnostics

```

pabs3, 2021-04-27:
For now I'm working around this crash using @inflex's [ideviceunback](https://github.com/inflex/ideviceunback) project.

nikias, 2021-04-27:
@pabs3 unfortunately Apple broke it some time ago, it's not "our" fault.

pabs3, 2021-04-27:
I'm definitely not blaming this issue on libimobiledevice, it is clearly a bug in `BackupAgent2`, but it should be easy to workaround that bug if there were a way to skip files.