libimobiledevice/libimobiledevice #1151: ERROR: Could not connect to lockdownd: SSL error (-5) after recent upgrades to ios 14.5 and macos 11.3

erlichg, 2021-05-01:
Hey guys,

So everything was working fine up until recent upgrades.
Not sure which one was the culprit as they were both done together.
Now I'm getting ERROR: Could not connect to lockdownd: SSL error (-5) for all idevice commands
Tried searching but all solutions were to clean/chmod/... the /var/db/lockdown folder and it is not possible as it is protected by SIP.
Tried reinstalling usbmuxd. All with latest versions.
Only thing that works is the simple version connection of ideviceinfo.
`
$ /usr/local/Cellar/libimobiledevice/1.3.0_1/bin/ideviceinfo -u XXXXX (hidden)

ERROR: Could not connect to lockdownd: SSL error (-5)
`

mexmer, 2021-05-03:
try 
idevicepair unpair 
then 
idevicepair pair

might be, that lockdown records inside phone got invalidated

erlichg, 2021-05-03:
Yeah. Already tried that. Forgot to mention.
Interestingly, it successfully unpairs and pairs, but list returns nothing and verify fails with same error

mexmer, 2021-05-03:
will check that later, i have one phone with 14.5 beta, other devices on 14.4 still