libimobiledevice/libimobiledevice #1155: idevicebackup2 unback problem

DJManas, 2021-05-05:
Hello,

it might be mi dumbness, but I have problem. I have iOS 14.4.2 and I want to extract one file from backup, so I use:
```
~/iPhoneBackups $ idevicebackup2 backup --full c/
Backup directory is "a/"
Started "com.apple.mobilebackup2" service on port 49325.
Negotiated Protocol Version 2.1
Starting backup...
```

Everything goes well, but when I want to unback the backup, it have problem that I don't know password. I had never backed this iphone to iTunes, only iCloud. Also I had tried my iCloud password for encryption off, but no luck. Sometimes it asks me for entering the pin on my phone, but then it still fails to do something.

idevicebackup2 -i unback c/
```
Backup directory is "c/"
This is an encrypted backup.
Enter backup password: *******************
Started "com.apple.mobilebackup2" service on port 49349.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting to unpack backup...
Backup password: Yes
Sending 'uid/Manifest.plist' (174.6 KB)
ErrorCode 207: Invalid password when restoring encrypted backup (MBErrorDomain/207). Underlying error: MKBKeyBagUnlock error (-3) (MBErrorDomain/207).
The files can now be found in the "_unback_" directory.
Unback Successful.
```

idevicebackup2 -i encryption off
```
Started "com.apple.mobilebackup2" service on port 49354.
Negotiated Protocol Version 2.1
Enter current backup password: *******************
Please confirm disabling the backup encryption by entering the passcode on the device.
ErrorCode 207: Invalid password (MBErrorDomain/207)
Could not disable backup encryption.
```

When I add directory it asks me for pin on iphne, but no luck
idevicebackup2 -i encryption off c/
```
Started "com.apple.mobilebackup2" service on port 49378.
Negotiated Protocol Version 2.1
Please confirm disabling the backup encryption by entering the passcode on the device.
ErrorCode 207: Invalid password (MBErrorDomain/207)
Could not disable backup encryption.
```

I had pulled the last version of libimobiledevice today 5th of May and usbmuxd had no updates. What am I doing wrong?
Thanks,
Regards,
Petr Sourek

nikias, 2021-05-05:
@DJManas did you maybe set up your device with a Mac? Chances are that your password is stored in the keychain. If not then you're out of luck. In any case, the backup password needs to be set by the user through a connected computer so this is not something that happens by accident, and is also not set directly on the device.

DJManas, 2021-05-06:
@nikias Thanks for the info. The iPhone was never connected to PC regardless operating system it was setup to sync only with iCloud. The problem is, that my MBA drive died and the battery in iphone is getting swolen, so before I repair it, I wanted one sqlite database from the iphone to get out and I am on Linux at the moment.

Never mind, thank for your reply. I am closing this issue, will try to install Windows into virtualbox and get the file from there and then repair it.

closed 2021-05-06 by DJManas

mexmer, 2021-05-06:
if you have access to phone, you can create new backup, if you reset phone settings, but of course it will not help with decrypting already taken backup. 
https://support.apple.com/en-us/HT205220

mexmer, 2021-05-06:
oh, one more thing regarding itunes backup password, if you do restore backup from "old" phone, to "new" phone, it retains backup password, it also does it, if you use apple trasnfer utility (when setting up new phone).

when i needed to service my ipad, i did do itunes backup to computer (with password), they gave me new one, because it was not serviceable, i restored to new my backup, then later i made backup from new one, and when i wanted to browse it, it asked for password - which i never se to new ... but it was the one i used on old ipad.

DJManas, 2021-05-06:
Well the phone was never connected to PC, but when I take backup with libimobiledevice I cant decrypt it either. If I use encryption off, add passcode on the screen it tells me the thing I had passed in first post.

To be honest I only need one sqlite file, from application of Kniha Jizd, which doesnt support sync to iCloud and thats it. Now I had mounted APFS TM backup and copying virtual machine. Will try to connect it to it and install itunes to get the file (it used to be in the applications in the iTunes and where you could upload files there).

But thanks for the info and definitelly thanks for this library it seems very helpful, but I am not sure if I will stay in Apple ecosystem for long to give good use to it.