libimobiledevice/libimobiledevice #1157: M1 macbook support for UDID and Info

shadowwalkersteam, 2021-05-08:
Is there is any way to get the UDID or info of the M1 macbook connected with another mac via USB-C cable? I was able to restore the M1 mac via apple configurator and I noticed during the restore process it throw a UDID when I ran idevice_id on it but after mac restored and boot up it doesn't give any UDID. 
My goal is to fetch the device info of M1 Mac like we do for iOS devices.

nikias, 2021-05-08:
It is only visible as USB device in restore mode.

shadowwalkersteam, 2021-05-09:
Is there is any way to fetch its device info?
Also when it get restore successfully then it show the correct serial number in the USB List of macbook before that it only shows CPID BDID combined with some other data in the serial number.

nikias, 2021-05-09:
First, does "the correct serial number" refer to the UDID or the serial number?
Then, what "other" data do you get? The UDID of the macbook (and new iPhones) composed of CPID + ECID, so if you have those you have the UDID.

shadowwalkersteam, 2021-05-09:
Yes after M1 mac finishes the Restore, it shows the correct serial number (referring to Serial number instead of UDID), before Restore it doesn't even detect and when we send it to DFU mode there it shows CPID etc and ECID and I use ECID to flash it, once flash finishes all that CPID ECID replaced with the Original Serial Number of the mac. 
So my point is I don't get the device info and even if i try idevice_id it doesn't print anything so I was concerned How i can get the full device info if idevice_id is not showing any UDID.

nikias, 2021-05-09:
`idevice_id` can only print it if `usbmuxd` is showing the device, and the mac in normal mode is not a usbmux device, so it won't show up. Also `usbmuxd` and thus `idevice_id` does not show devices in recovery or DFU mode.
But well, in DFU mode you get the required info, so what are you missing?
```
sprintf(uuid, "%08x-%016llx", CPID, ECID);
```

shadowwalkersteam, 2021-05-09:
I want to get some more info like Storage capacity etc and RAM and Part number etc etc.

shadowwalkersteam, 2021-05-09:
But I don't understand one thing if mac flash finishes why it shows the correct serial number when i go to system report of the Mac and click on USB there it shows correct serial number, from where it comes after flash, there must be something which fetches the serial number of the mac after flash finishes and mac boot up

nikias, 2021-05-09:
Possibly if connected during boot, even without a restore, it will be visible, that's my guess.
As for `ideviceinfo` that will probably never work as that's not how Apple decided to do things.

shadowwalkersteam, 2021-05-09:
I tried with Apple configurator 2 and it wasn't showing any info either except the ECID even apple configurator wasn't showing the serial number too. 
Don't if it will be possible to get the detailed info of the M1 mac.