libimobiledevice/libimobiledevice #1160: SysCfgData gets empty

5645q, 2021-05-13:
SysCfgData gets empty  domain:"com.apple.mobile.internal" key: "SysCfgData"

ZoMinster, 2021-06-21:
iPhone 12 is empty, there is value  < iPhone 12 ，do you know how to extra SysCfgData ?

truonggiang0710, 2021-06-24:
Yes, I have the same. Do you have any idea @ZoMinster @5645q ?

G-Yong, 2022-01-04:
"SysCfg" gets empty  with function <diagnostics_relay_query_mobilegestalt()> too.
Someone said there is a new key(or method),  sadly, he is unwilling to tell me.

Johnny314, 2022-03-25:
Hi any one know how we can fetch SysCfg on 12 or 13 series

Johnny314, 2022-07-30:
@G-Yong  @truonggiang0710 @ZoMinster  are you able to fetch Syscfg on newer devices?

G-Yong, 2022-08-01:
> @G-Yong @truonggiang0710 @ZoMinster are you able to fetch Syscfg on newer devices?

Not yet, I'm still working on it.

murtz53, 2023-01-04:
Any luck with SysCfg / SysCfgData on iPhone 12 and above?

truonggiang0710, 2023-08-08:
is there any update on iPhone 12 and above @murtz53 @G-Yong ?

truonggiang0710, 2023-08-10:
> "SysCfg" gets empty with function <diagnostics_relay_query_mobilegestalt()> too.
> Someone said there is a new key(or method), sadly, he is unwilling to tell me.

Did you have any solution to do this? @G-Yong

AiXanadu, 2024-04-07:
@ZoMinster @truonggiang0710 @G-Yong @murtz53 @Johnny314 Hi, is there any progress?

AiXanadu, 2024-11-14:
> “SysCfg” 在函数 <diagnostics_relay_query_mobilegestalt（）> 中也变为空。有人说有新的密钥（或方法），遗憾的是，他不愿意告诉我。

有知道是什么软件吗？或许可以逆向。

AiXanadu, 2024-11-16:
> > @G-Yong @truonggiang0710 @ZoMinster are you able to fetch Syscfg on newer devices?
> 
> Not yet, I'm still working on it.

Is there any latest progress? I am willing to pay to get the method.