libimobiledevice/libimobiledevice #1161: Support for listing running processes

nisargjhaveri, 2021-05-14:
How can I list running processed on a device? The pid is required to attach to an existing process on device, but I could not find a way how I can find the pid. 

Xcode is able to list the processes on the device. So I guess it should be possible. 

Any suggestions on where to look? 

Related: https://github.com/libimobiledevice/ideviceinstaller/issues/87

yacosdad, 2021-08-02:
Hi, have you resolved this problem by any other means?

nisargjhaveri, 2021-09-15:
I found this: https://github.com/troybowman/ios_instruments_client

This tool can list out running process along with some other things. Though only on Mac.

It essentially talks to `com.apple.instruments.remoteserver` service to list of running processes.

mexmer, 2021-09-15:
protocol seems simmilar to debugserver protocol
so might be doable

tool you linked talk either to
```
com.apple.instruments.remoteserver
```
or
```
com.apple.instruments.remoteserver.DVTSecureSocketProxy
```
i suspect later will be required on ios 14

it sends command
```
_requestChannelWithCode:identifier:
```
with agument
```
com.apple.instruments.server.services.deviceinfo
```

which causes proxyservice inside device to switch to different service module
then it sends command
```
runningProcesses
```

i assume this can be also implemented with libimobiledevice, unless something is missing in debugserver interface

nisargjhaveri, 2021-09-15:
I believe the com.apple.instruments.remoteserver* services uses DTXMessage to communicate, which is not the case in debugserver. That might be something we might need to add in libimobiledevice, if I'm correct.

mexmer, 2021-09-15:
@nisargjhaveri yeah, you are right, my bad
seems that guy also made separate klient for DTXMessage
https://github.com/troybowman/dtxmsg